usnistgov/dastard #5: File writing

ggggggggg, 2018-05-25:
implement ljh3 and "OFF" file writing

joefowler, 2018-05-25:
Keep in mind that we haven't decided what file format to use for projection-only data. On the table are:

1. HDF5 (hope dastard doesn't crash!)
1. Dirfiles (see [my incomplete port](http://github.com/joefowler/gogetdata) of [GetData](http://getdata.sourceforge.net)). Would offer sweet KST for plotting and histogramming, but still have the problem that _either_ each TES channel is a distinct dirfile _or_ KST will have to pad each channel's pulse height or energy list with zeros to match the lengths of all lists. Ugh.
1. LJH, or a perversion of it.
1. ASDF, which sounds great but doesn't exist really, and especially not in golang. See [this binding](https://github.com/astrogo/asdf), which has not been touched in over 2 years.
1. A roll-your-own binary format. 

While this last one might sound fine, I have a strong feeling that storing metadata with the data would lead to considerable reduction in pain down the road. LJH stores metadata, but in some completely ad hoc, barely specified way. We could expand it, but only by US doing work. Just don't immediately shut down the idea of using GetData, or gogetdata.

ggggggggg, 2018-06-29:
im going to close this in liu of more specific issues, we have #9 available as a reminder, and LJH3 writing is in dastard, but not in dastard commander

closed 2018-06-29 by ggggggggg