usnistgov/dastard #6: How to do projections on variable length records?

ggggggggg, 2018-05-25:
The simplest option is simply keeping a list of projectors for different record sizes, and choosing the appropriate one in each case. It's also possible we should be whitening the data instead, which 
I believe would allow truncation of the projectors.

joefowler, 2018-06-04:
See my email to you dated June 1. This is complicated, and I think we should defer decisions until we know more. Let's solve this problem only after we know what we're solving.

ggggggggg, 2019-08-06:
I think Dastard should just start writing variable length records to disk. Even though we don't know how to use them, they're easy to ignore, and it will be easier to learn how to use them if we can just pick and choose datasets to look at since they all have that info. I propose the following. Also I think it will be easier 
1. For LJH 22 files, Dastard writes a trace of all zeros or some other sequence unlikely to be observed in reality. Or I could write the actual data, and zeros otherwise, or is there an invalid number i can choose? Or I could just not write them. The reason to write them is to preserve the 1-1 mapping between pulses in the OFF file to pulses in the ljh file.
2. For OFF files/real time analysis I just truncate the projectors and project with them. Sure it's not the best possible thing to do, but it's more information than not doing it.
3. LJH3 is still unused, but should already work in this case.

I think it will be pretty easy to change what projectors we use for shorter records later.