usnistgov/dastard #10: go test -race fails

ggggggggg, 2018-06-05:
This might indicate a real problem not sure.
```
PN101133:dastard oneilg$ go test -race
==================
WARNING: DATA RACE
Read at 0x00c4201af210 by goroutine 62:
  github.com/usnistgov/dastard.TestTriangle()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/simulated_data_test.go:75 +0xbf0
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:777 +0x16d

Previous write at 0x00c4201af210 by goroutine 81:
  github.com/usnistgov/dastard.(*DataStreamProcessor).ConfigurePulseLengths()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/process_data.go:99 +0xb0

Goroutine 62 (running) created at:
  testing.(*T).Run()
      /usr/local/go/src/testing/testing.go:824 +0x564
  testing.runTests.func1()
      /usr/local/go/src/testing/testing.go:1063 +0xa4
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:777 +0x16d
  testing.runTests()
      /usr/local/go/src/testing/testing.go:1061 +0x4e1
  testing.(*M).Run()
      /usr/local/go/src/testing/testing.go:978 +0x2cd
  github.com/usnistgov/dastard.TestMain()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/rpc_server_test.go:166 +0x18c
  main.main()
      _testmain.go:92 +0x22a

Goroutine 81 (finished) created at:
  github.com/usnistgov/dastard.(*AnySource).ConfigurePulseLengths()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/data_source.go:211 +0xdc
  github.com/usnistgov/dastard.(*TriangleSource).ConfigurePulseLengths()
      <autogenerated>:1 +0x5f
  github.com/usnistgov/dastard.TestTriangle()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/simulated_data_test.go:72 +0xb73
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:777 +0x16d
==================
==================
WARNING: DATA RACE
Read at 0x00c4201af218 by goroutine 62:
  github.com/usnistgov/dastard.TestTriangle()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/simulated_data_test.go:75 +0xed6
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:777 +0x16d

Previous write at 0x00c4201af218 by goroutine 81:
  github.com/usnistgov/dastard.(*DataStreamProcessor).ConfigurePulseLengths()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/process_data.go:100 +0xcb

Goroutine 62 (running) created at:
  testing.(*T).Run()
      /usr/local/go/src/testing/testing.go:824 +0x564
  testing.runTests.func1()
      /usr/local/go/src/testing/testing.go:1063 +0xa4
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:777 +0x16d
  testing.runTests()
      /usr/local/go/src/testing/testing.go:1061 +0x4e1
  testing.(*M).Run()
      /usr/local/go/src/testing/testing.go:978 +0x2cd
  github.com/usnistgov/dastard.TestMain()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/rpc_server_test.go:166 +0x18c
  main.main()
      _testmain.go:92 +0x22a

Goroutine 81 (finished) created at:
  github.com/usnistgov/dastard.(*AnySource).ConfigurePulseLengths()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/data_source.go:211 +0xdc
  github.com/usnistgov/dastard.(*TriangleSource).ConfigurePulseLengths()
      <autogenerated>:1 +0x5f
  github.com/usnistgov/dastard.TestTriangle()
      /Users/oneilg/go/src/github.com/usnistgov/dastard/simulated_data_test.go:72 +0xb73
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:777 +0x16d
==================
--- FAIL: TestTriangle (0.01s)
	testing.go:730: race detected during execution of test
FAIL
exit status 1
FAIL	github.com/usnistgov/dastard	3.689s
```

joefowler, 2018-06-05:
Well, I've run the tests with -race before, too, and I did get failures. I also don't know what it means! Let's leave this open until we've merged our various threads of work and see to it then.

joefowler, 2018-06-19:
I don't know if we have fixed every possible race, but the race seen by `github.com/usnistgov/dastard.(*DataStreamProcessor).ConfigurePulseLengths()` is fixed in 800180ca by locking the  RPC server calls and the publication of the resulting status.

closed 2018-06-19 by joefowler