usnistgov/dastard #12: Our ZMQ binding is unsupported

joefowler, 2018-06-11:
I'm not sure how I missed this giant warning in the [ZMQ binding's REAME](http://github.com/alecthomas/gozmq):

    NOTE: These gozmq bindings are in maintenance mode. Only critical bugs will be fixed. 
    Henceforth I would suggest using @pebbe's actively maintained bindings for zmq2, zmq3 and zmq4.

Anyway, we need to use [pebbe's zmq4](https://github.com/pebbe/zmq4) package instead. The current binding works on Macs but won't build on our Linux machines. I just figured this out today. Duh!

joefowler, 2018-06-11:
Uh, hold on. We are NOT using the binding from Alecthomas. I misunderstood.

* Not using: [gozmq](https://github.com/alecthomas/gozmq), which is deprecated
* We are using [czmq](https://github.com/zeromq/goczmq), which isn't. Also, Galen says he compared it to how [pebbe's zmq4](https://github.com/pebbe/zmq4) worked. They were similar.

CHANGE this issue: we cannot get goczmq to pass tests on the Raven computer.

joefowler, 2018-06-11:
My bad. The problem was that ubuntu 16 does not support CZMQ 4.x, which the [goczmq](https://github.com/zeromq/goczmq) package requires. Here are my notes on getting Ubuntu 16.04 LTS to use CZMQ 4:

Need CZMQ version 4 to work with goczmq. How to get it on Xenial (Ubuntu 16.04 LTS), given that the system czmq is 3.0.2-5?

See [this Stack Overflow question](https://stackoverflow.com/questions/26020917/what-is-the-function-of-etc-apt-sources-list-d) about adding an apt-get source for more on apt sources and their keys.

First, add a file to /etc/apt/sources.list.d/zeromq-stable.list containing this line: 

```text
deb http://download.opensuse.org/repositories/network:/messaging:/zeromq:/release-stable/xUbuntu_16.04 .\
```

Substitute "git" for "release" and "draft" for "stable" as desired. That won’t be enough, because you don’t have the repository's key, which you’ll notice if you try to run `sudo apt-get update` or `synaptic`

```bash
wget -O - http://download.opensuse.org/repositories/network:/messaging:/zeromq:/git-draft/xUbuntu_16.04/Release.key | sudo apt-key add -
sudo apt-get update
sudo apt-get synaptic 
```

And then uninstall/install the obvious things.

closed 2018-06-11 by joefowler

ggggggggg, 2018-06-11:
Here are the equivalent commands on Ubuntu 14, note there is a command that doesn't require you to manually modify a file `sudo add-apt-repository`

```
sudo add-apt-repository ppa:ondrej/php -y   # for libsodium
sudo apt-get -qq update
sudo apt-get install -y libsodium-dev libzmq3 libzmq3-dev

sudo add-apt-repository 'deb http://download.opensuse.org/repositories/network:/messaging:/zeromq:/git-draft/xUbuntu_14.04/ ./'
wget http://download.opensuse.org/repositories/network:/messaging:/zeromq:/git-draft/xUbuntu_14.04/Release.key
sudo apt-key add - < Release.key
sudo apt-get install libczmq-dev
```