usnistgov/dastard #13: How to publish full records?

joefowler, 2018-06-14:
@ggggggggg your recent work from Japan on publishing and saving data to files has broken the ability to plot records in microscope.

Is this to be expected? Is there something that I need to set up properly? I'll study `publish_data.go` and see if I can figure it out. I just didn't expect that publication of triggered records should have been disabled by default.

If it was, then do we want to keep it that way?

ggggggggg, 2018-06-14:
I think it's pretty reasonable to enable it by default. I just wanted to keep the API the same as the other publishing methods. Might as well enable the summaries by default as well.

All you should have to do is call `DataPublisher.SetPubRecords()` and `DataPublisher.SetPubSummaries()`. Maybe do it in `cmd/dastard` so it's not baked into the library?

joefowler, 2018-06-14:
Well, I am at a complete loss when it comes to deciding what goes in "the library" and what goes in `cmd/dastard`. I tried to keep it minimal in the command, but I did need to handle the finding of configuration file(s).

If keeping the API the same for publishing to ZMQ and "publishing" to files is the concept, then maybe activating and deactivating PubRecords and PubSummaries is something that ought to be controlled over RPC? But what happens if one client deactivates them when another was using them? Dastard could keep a count of "clients who want this feature", but that won't work, because clients can silently disappear.

In order to stay on target for this week, I'm going to activate both by default.

joefowler, 2018-06-14:
I could imagine a real performance hit for trying to publish every channel, every pulse and every summary. Sure, ZMQ filters packets at the server side if there aren't any subscribers, but there is the problem that we still have to convert every record to a message before ZMQ gets a chance to filter.

New plan: activate both for this week's test. Then later _if needed_, we can undo that. We'd require that all clients who need records or summaries must regularly re-request them for the channels that they want. Some kind of monitor in dastard would shut off the publication of each channel's records/summaries if no request has been made in the last, say, 30 seconds. That would add complexity to `microscope`, unfortunately, because it currently has no need to connect to the JSON-RPC server. 

Oh! If microscope is launched from the DC gui, then DC can turn on and off publishing? But not really, because DC doesn't know which channels are being requested. Ugh. Boy, it will be far easier if we can just publish everything and hope it doesn't bog down the system too much. Let's optimize this later if needed only.

ggggggggg, 2018-06-14:
Lets wait till we see a performance problem. I measure about 4 us/(1000 sample record) published over zmq, so hopefully we don't see a problem until at least 200k records/second. Ohh I see thats what you ended with.

I think you made an argument at some point that we're mostly interested in worst case performance, because we want to handle the worst case, and don't care much if we use too much cpu in easy cases. So I think we just start with what is easy, and work on bottlenecks as they occur.

joefowler, 2018-06-14:
For RAVEN, we anticipate 100-400 records per second, times 240 TESs. I think this is the regime where we can hope--even expect--that the workload (for the computer) is tolerable, but we cannot be certain in advance. 

The test you mentioned: was that publishing without a subscriber, where ZMQ's server-side filter drops all (or most) packets? That is probably the typical case we have to worry about.

One possible way to improve performance (if needed): throttle back on the messages. The plotter is set up not to plot any new records for 0.5 (or 0.75?) seconds after the last new plot. If we decide not to use these full-record messages for any other purpose, then we can filter them _before_ they are made into messages. If we decide to split file-writing out of dastard, then this isn't a possibility, but we are already well along a path where file-writing is in dastard. Are there other purposes to the PubRecords port, real or anticipated?

I don't know if I was the one who said it, but it is a sensible viewpoint: we need to optimize dastard's worst-case performance, and we can tolerate inefficiency in all other cases.

joefowler, 2018-06-14:
Fixed in 27b8012.

closed 2018-06-14 by joefowler