usnistgov/dastard #14: Lancero with testing

joefowler, 2018-06-14:
I'm making good progress at getting the lancero driver and dastard to talk to one another. The system works, as far as I can tell. However, there's a lot of code that is now untested. Make sure that the `lancero` package AND the `lancero_source.go` file both get extensive tests before we declare complete success.

joefowler, 2018-06-16:
#19 covers the `LanceroSource` object to talk to the driver. But I still haven't figured out how to test it! Not even provisionally, i.e. on a system that has an available card. It should be possible, though.

joefowler, 2018-06-25:
See #26. This is an attempt to test Lancero sources by providing an interface and a no-hardware implementation.

ggggggggg, 2018-06-29:
fixed by #26

closed 2018-06-29 by ggggggggg