usnistgov/dastard #23: Closing of a closed channel. Race condition?

joefowler, 2018-06-19:
You sometimes see a `panic: close of a closed channel` and sometimes don't during testing. Is this a race condition? Here's an example output, from the Travis CI run, build 177.2 on Go 1.10 (but I restarted that build, so the evidence has vanished). A sample from the output:

```text
=== RUN   TestTriangle
panic: close of closed channel
goroutine 90 [running]:
github.com/usnistgov/dastard.(*AnySource).CloseOutputs(0xc42024e0f0)
	/home/travis/gopath/src/github.com/usnistgov/dastard/data_source.go:202 +0x4a
github.com/usnistgov/dastard.Start.func1(0xa9cd80, 0xc42024e0c0)
	/home/travis/gopath/src/github.com/usnistgov/dastard/data_source.go:68 +0x8e
created by github.com/usnistgov/dastard.Start
	/home/travis/gopath/src/github.com/usnistgov/dastard/data_source.go:59 +0xef
FAIL	github.com/usnistgov/dastard	0.880s
?   	github.com/usnistgov/dastard/cmd/dastard	[no test files]
?   	github.com/usnistgov/dastard/cmd/vipertest	[no test files]
=== RUN   TestFromGetBytes
```

ggggggggg, 2018-06-19:
I saw this approximatley every other save last night.

joefowler, 2018-06-19:
I see it on my machine, but much less often than half the time. That's why I haven't made this an, um, issue until now. Perhaps the race detector will help me...

joefowler, 2018-06-25:
I haven't failed any `go test -race` lately, and I do try often. So I'll close this issue.

closed 2018-06-25 by joefowler