usnistgov/dastard #30: Microscope plots of error channels are wrong b/c assume unsigned data

joefowler, 2018-06-25:
Hey @ggggggggg: 

What the heck should we do about the fact that dastard -> microscope currently plots error signals as if they are of the usual `RawType` (defined as `uint16`), but error signals are actually meant to be interpreted as _signed_ integers?

Keep in mind that we have in microscope a sense of “raw units” and “physical units”. This is about trying to get the _raw_ units correct (I think physical will follow if we do this).

Ideas:

1. Add to the dastard `DataRecord` type a bool to indicate signedness. Set to true for error channels.
2. Add to the dastard `DataRecord` type a value “offset” of type `RawType` to signal the offset of the zero value. Set to 32768 or whatever for error channels. (Probably a needless generalization of 1.)
3. Actually add 32768 or whatever to all the values in the `[]RawType` that hold the data. In this way, values near signed-zero get moved away from zero. (Probably would work but is unnecessarily confusing, because the raw data that you see aren’t the actual underlying raw data.)
4. Create a variant DataRecord where the data item is of type `[]RawSignedType` instead of `[]RawType`. (Go makes this kind of impossible, because it doesn't have union-like types.)
5. Ignore in Dastard. Fix only in microscope. (I think this is impossible, because Microscope won’t have any way to know which channels need data to be considered signed (not unless you implement a per-channel bool to indicate signedness, and that’s idea 1.)

Keep in mind that it's not just a microscope problem: the data summary values are probably crap for error channels until we fix this. Do we care?? Should we be computing them for error channels?

joefowler, 2018-06-26:
Note that the volts per arb scaling is probably something we ought to include in the `DataRecord` object. I'm thinking we need to revise that object to including both the volts/arb and the signed boolean, as in my idea 1 above. Yeah, it's repetitive, but boy is that easier than maintaining that state elsewhere. (While I'm at it, I might reduce some numbers in that object from `int` and `float64` to `int32` and `float32`.)

With these two facts embedded in each `DataRecord`, I would be able to add them to the message produced in `func messageRecords` and `func messageSummaries`.  Oh, duh! They are already in these messages, but they are filled with default values instead of true values.

So I think I see the right path. Once I follow it, microscope can properly plot data; _and_ also the analyzed quantities can be computed correctly whether the data are signed or not.

Let me know if you have strong feelings in favor of some other approach.

ggggggggg, 2018-06-26:
I like any idea that makes the error signals more workable, my experience with MATTER/NDFBServer/Mass was that error signals just didn't work right, and it was annoying a few times.

Also it seems like the signedness of the error signal should matter for the mix.

joefowler, 2018-06-26:
Oh yeah, the mix. Good point! So I will attempt idea 1. To do items are:

- [ ] Store signedness and scaling-to-volts in the `DataRecord`
- [ ] Fix the published messages in `func messageRecords` to use signedness and scaling
- [ ] Fix the mix to account for signedness
- [ ] Fix the `DataStreamProcessor.AnalyzeData` method to account for signedness.

I'll take care of them.

commit 618006bf (2018-06-26): Edge and level triggering handles signed data values

Fixes #30. I think...
closed 2018-06-27 by joefowler