usnistgov/dastard #39: Timestamps might be more irregular than they need to be

joefowler, 2018-06-27:
Auto triggering at 250 ms suggests that either the auto-trigger algorithm is not regular, or (more likely) that the time stamping algorithm is wrong. Is there a long latency, or is the wrong time being assigned? Or is there an error in the computation of the first sample's timestamp from the last sample's timestamp?

joefowler, 2018-06-29:
This is confirmed by looking at LJH files. But they aren't easy to read, because something about their new format is screwing up MASS. Ugh!

One thing I can say is that the _mean_ time between auto-triggered records is indeed the correct value of 250 ms. But the standard deviation is 40 ms, and the max and min are 357 ms and 150 ms. On the row-counter side, the rows are strictly 74,404 counts apart each time. It's a 2x30 TDM array with LSYNC of (I thought) 30. The file claims that the timebase is exactly 7 microseconds, though I believe that ought to be 7.2 microseconds. 

In other words:

1. The auto-triggering is happening at a very regular interval, according to the row counter.
1. The auto-triggering timestamp is very irregular.

joefowler, 2018-07-07:
After lots of work, this issue is not totally resolved, but it is clearly improved on simulated sources. It will need careful study on Lancero sources. Next week...

joefowler, 2018-07-10:
Commit 18e80b1 tried to address this, but it offered no improvement whatsoever.  I'll have to think it over some more.

ggggggggg, 2018-07-10:
Probably you've already looed, but this is where it is handled in NDFBServer. If remember correctly, we got to the point where the difference between the timestamp difference and the rowcount difference was usually below 20 us. Though I don't remember the exact test details.

https://bitbucket.org/nist_microcal/nasa_daq/src/3eed1a3bb8c1e0270d3a2b8201fa19e9beeba000/ndfb_server/NDFBCardPCIe2013.cpp#lines-136

joefowler, 2018-07-10:
Yes, I am experimenting with new ideas. It's clear that I was not well approximating what was done in the C++ code, but my first attempt at approximating it better (18e80b1) did not improve the results at all! I'll try again. I have a few ideas, one of which is that we might be getting the "time fix" correctly but are somehow incorrectly subtracting the time to get from the time fix to the time-of-first-sample.

commit e02358f8 (2018-07-10): Fix timestamps for lancero sources

Fixes #39
commit 90eea045 (2018-07-11): Fix data timestamps for sim sources, too

Fixes #39
PR #64 (2018-07-11): Fix lancero timing
Commit e02358f fixes #39 this in grand fashion for the Lancero source. In my test case, there are 1375 records taken with 250 ms pulse auto-triggers, so over 5 minutes of data.  Every record is 34722 frames apart, as expected (7.2 µs timebase, so 34722 frames = 249.9984 ms).

But the cool part is that the mean time between records' timestamps is 250.003 ms with a standard deviation of 3.77 µs. Hooray!

While the standard deviation for simulated pulse sources is more like 50 µs, we also don't really care.

One problem with this PR is that it doesn't have a test to exercise th [... cut]
closed 2018-07-11 by joefowler