usnistgov/dastard #48: Lancero NSAMP should be known to Dastard; mix should use it.

joefowler, 2018-06-29:
If Dastard doesn't know the Cringe "NSAMP" value when we have a Lancero source, then we cannot compute the correct conversion between Arbs and Volts for error signals. This is a minor point but also not super hard to fix.

Also, as it stands, any _change_ in NSAMP will invalidate the mix parameters. For this problem, we have two possible solutions:

1. Ignore this problem. Don't change NSAMP without next running autotune. The new mix parameters will be scaled to the new value of NSAMP. Problem solved.
1. Use NSAMP inside Dastard to (conceptually) convert Error _sums_ into _means_. If we do this, then the same mix parameters computed at one NSAMP value ought still to work at another value. Of course, that assumes that the user is sharp enough to update the NSAMP value when she or he changes it in Cringe.

Discuss.

ggggggggg, 2018-06-29:
I just noticed there is a variable named `nsamp` used in some of the triggering code as shorthand for `NSamples`. Might be a good idea to rename that when we add a meaningful `NSAMP`.

joefowler, 2018-06-29:
Okay, I believe that we agreed to the following:

* GUI users set the NSAMP value, as with ndfb_server.
* The value is sent to Dastard as a configuration item and stored (and sent back to other clients).
* The value is used inside Dastard to correctly compute Volts/arb for error channels.
* The value _might be_ used in Dastard to rescale the error values when computing the mix, but _only_ if this is how ndfb_server does it OR if this is how autotune will work in the future. (I'd bet that we can't easily change autotune to work one way for ndfb_server and another for dastard, so I doubt we want to change it.)

joefowler, 2018-07-05:
Notice in the following excerpt of `NDFBServer.cpp` that the mix level that we work with is supposed to be divided by "co-added samples" before being multiplied by an error signal. That is, the "mix level" is to be multiplied by the _mean error value_ whereas the raw data contains the _sum of error values_.

```c++
                if (fbStream[row]->mixFlag())
                {
                    float mixLevel = fbStream[row]->mixLevel();
                    if (fbStream[row]->mixInversionFlag())
                        mixLevel *= -1;
                    mixLevel /= coAddedSamples();

                    // Since "mix" mixes error[k] into the 1-sample-older feedback, we have
                    // shifted the rawFBSignals[] vector so that rawErrorSignals[k] and
                    // rawFBSignals[k] are the pairs that are to be mixed together.
                    for (int j=0; j<nvalues; j++)
                    {
                        // CAREFUL!  Use the round() function, else the mix will truncate towards zero
                        // That makes zero an "attractor" for all (mixLevel*error) between -0.999 and 0.999.
                        // That is twice the range of any other mixed-in integer.  Boo!
                        double mixed =  rawFBSignals[j] + round(mixLevel * (int16_t) rawErrorSignals[j]);

                        // Make sure that mix doesn't push FB out of the range 0-65535 (=UINT16_MAX).
                        if (mixed<0)
                            rawFBSignals[j] = 0;
                        else if (mixed > UINT16_MAX)
                            rawFBSignals[j] = UINT16_MAX;
                        else
                            rawFBSignals[j] = mixed;
                    }
                }
```

I will make Dastard do the same thing.

commit a3669736 (2018-07-06): Rescale mix fraction by 1/NSAMP

If you request a channel have mix fraction of 1.0 but its NSAMP=4, then the effective multiplier of the error signal is 0.25. Fixes #48
closed 2018-07-06 by joefowler