usnistgov/dastard #66: Dastard won't shut up with the NUMBERWRITTEN messages

joefowler, 2018-07-11:
Set up a real or simulated data source. Start writing and then stop writing.

The server keeps emitting pointless NUMBERWRITTEN messages. I would like to see these messages stop when file writing stops. If they do, then the text on d-c's writing tab will tell you how many pulse records were written in the immediately previous file.

ggggggggg, 2018-07-11:
Yeah, so the reason it is like this is these lines in `PublishData`

```go
	if (dp.HasLJH22() || dp.HasLJH3() || dp.HasOFF()) && !dp.WritingPaused {
		dp.numberWritten += len(records)
	}
	if dp.numberWrittenChan != nil {
		dp.numberWrittenChan <- dp.numberWritten
	}
```

Before I had something like
```go 
	if (dp.HasLJH22() || dp.HasLJH3() || dp.HasOFF()) && !dp.WritingPaused {
		dp.numberWritten += len(records)
		dp.numberWrittenChan <- dp.numberWritten
	}
```
instead, which behaves as you request. But It would hang when I turned on an off file writing, probably because not some channels sent a message to `dp.numberWrittenChan` and others did not. The goroutine that reads from `dp.numberWrittenChan` loops over all channels and gets mad if they aren't all there.

joefowler, 2018-07-11:
We just discussed this on the phone. We decided that you would add an unexported field `writingActive bool` or something. I'm not sure how to set it correctly, but then in `function (ps *PublishSync) Run()` you could test it and only publish the NUMBERWRITTEN message if it's true.

joefowler, 2018-07-11:
Fixed in da03189

closed 2018-07-11 by joefowler