usnistgov/dastard #69: We still have race conditions, now in the data publishers

joefowler, 2018-07-12:
I'll look into it, but I mean that you can run `go test -race` to see.

joefowler, 2018-07-12:
Actually, I've been toying with mutex use to try to prevent these races, and I'm not even sure it's possible. Or if it's possible, it might be a lot of confusing work. We have 3 choices that I can see:

1. Continue trying to prevent any data races by clever, if not omniscient, use of `sync.Mutex`, popping it in one place at a time as needed. There would be a lot of deadlocks and difficult-to-debug problems.
2. Give up and accept that our architecture requires a lot of data races. Hope that they are all of the benign sort, where we never care whether we read before or after a value has been changed, and we never get unlucky and read inconsistent values. If we take this route, then we are ignoring the lessons of [Benign Data Races: What Could Possibly Go Wrong?](https://software.intel.com/en-us/blogs/2013/01/06/benign-data-races-what-could-possibly-go-wrong). See also [this paper](https://www.usenix.org/legacy/event/hotpar11/tech/final_files/Boehm.pdf).
3. Redesign, so that no goroutine changes any state required by another, not by setting variables directly, nor by calling methods that set them. I don't know how to do this. I guess each goroutine whose state can change needs it to change the same way we signal them to stop: it can loop forever with a select statement, and there can be a case (or the default) for quitting the loop, a case for doing its normal business, and a case (or several) for accepting state-change requests on a channel or channels.

I would hate to pursue idea 3, if it is simply letting the best become the enemy of the good-enough. But I would also hate to continue on path 2, lest it lead to mysterious failures and unreliable results down the line. Opinions?

ggggggggg, 2018-07-12:
I think 3 is the go way, from go/sync, emphasis mine. "Package sync provides basic synchronization primitives such as mutual exclusion locks. Other than the Once and WaitGroup types, most are intended for use by low-level library routines. **Higher-level synchronization is better done via channels and communication.**"

I agree it sounds a bit like letting the "best become the enemy of the good-enough", maybe just try to do future work in that style? Thats how I did `publishSync` and it wasn't too bad.

joefowler, 2018-07-12:
I haven't looked closely, but I'm surprised you say that `publishSync` was done this way, because we didn't fail the race test until you added OFF output to it!

One idea that's a mix between ideas 1 and 3 is:

4. Use Mutex locks to serialize reads and writes of state (e.g., trigger thresholds, output file objects, etc), as in idea 1. But idea 1 would require the "business end" of the worker goroutines to hold the lock while they do their work (or at least for the parts of their work where they need to read state like trigger thresholds). Instead, you could make a one-time, consistent copy of the state while holding the lock, and then release the lock and pass the consistent copy to the worker logic. This ensures that settings are always consistent _and don't change during a unit of work_.

This sounds kind of like a half-measure, and that idea 3 is still the way to go.

Perhaps sometime no earlier than mid-August, we can try for a redesign.

ggggggggg, 2018-07-12:
I took another shot at fixing the race condition with syncs, and it seemed like it worked until a totally unrelated test failed. I agree it's hard to do with mutexes.

Maybe each data publisher should have a `modifyChannel` that takes a giant `PublishState` object with all the argument needed to start publishing (including tings like nrow, ncol, filename, etc). I agree this shouldn't be a high priority, but its just annoying to leave it this way.

joefowler, 2018-07-13:
Well, before we actually try to finish this, we'll need some kind of survey of all the code. We would need to identify all objects that have either a "run loop" (such as `fund (ps *PublishSync) Run`) or an often-called "business function" (such as  `func (dp *DataPublisher) PublishData`).  

Then (to use idea 1) we'd have to remove all functions/methods that alter the internal state of these objects, replacing each with a channel that can carry an equivalent message to request the same state alteration. The pattern will be to handle all such pending requests on an object before repeating its run loop or starting its business function.

An important problem (see comments in #72) is that such channels will either have a finite capacity (and how do we pick it?), or zero capacity (and then they will block their senders).

I think this calls for more research. Perhaps a study of named [Concurrency Patterns](https://en.wikipedia.org/wiki/Software_design_pattern#Concurrency_patterns) would help us to identify a standard name for the problem we're trying to solve. It looks to me like the patterns [Active Object](https://en.wikipedia.org/wiki/Active_object), [Balking](https://en.wikipedia.org/wiki/Balking_pattern), [Reactor](https://en.wikipedia.org/wiki/Reactor_pattern) and [Asynchronous Method Invocation](https://en.wikipedia.org/wiki/Asynchronous_method_invocation) seem potentially relevant here.

ggggggggg, 2018-07-26:
When considering solutions, keep in mind that we generally should prefer low latency between command and effect to larger latency. How much to weight that is not clear.

ggggggggg, 2018-07-30:
I think this is a real issue still, I had a case where a call to `!dp.LJH22.HeaderWritten` failed with a nil pointer access, probably because `LJH22` was non-nil on the line before when I checked it, but was nil by the time I ran that. I think it mostly matters for publishing, so that should be the first place for focus fixes.

joefowler, 2018-09-25:
We've discussed a few times in September and agree that this problem needs to be addressed, by serializing the access to settings like trigger conditions, record lengths, etc. Idea 3 would be the only safe way to do it.

One problem that we've recognized is that the RPC service needs to be able to accept an unbounded number of requests during the time it takes to process a bundle of data and queue them up for handling after the bundle is processed (but before we have to start processing the next one). If there were a sensible bound on the number of requests, we could use a buffered Go channel, but there's no sensible way to choose that buffer size. Is 100 enough? 1000?

The way I imagined handling this is with a new object whose job is to queue up the requests in a slice (thus no limit on the number) and offer the requests in order on a channel to send them for handling when the time is right. Along these lines, the post [Building an Unbounded Channel in Go](https://medium.com/capital-one-developers/building-an-unbounded-channel-in-go-789e175cd2cd) might be relevant reading. The key insight that I see there is that for-select loops can mix channel reads and writes, doing whichever is possible. The other insight is that the job is a lot more complicated that you'd expect, since you have to handle empty queues that have nothing to send, and stuff like that.

joefowler, 2018-10-04:
See branch no_race_redesign (commit 4596f80) for the current status of this work. It's in good shape, but I just realized something thanks to the race detector: setting the Mix is still a race condition! Why? Because we don't *use* the mix settings during the process-data step like we use most other quantities. We use it in the produce-data step, and the new design allows produce-data (specifically, `LanceroSource.distrubuteData` does the de-MUXing and the mix) to run concurrently with handling RPC requests. Thus further work is needed to avoid *this* race.

joefowler, 2018-10-04:
Okay, I think 1c79504 fixes the problem with races in setting and using the mix levels.

Next: the races around `anySource.segments` which is shared by data producer and consumer goroutines. Replace them with a channel that carries a slice of segments?

joefowler, 2018-10-05:
Fixed for now, with PR #111.

closed 2018-10-05 by joefowler