usnistgov/dastard #78: Record length not restored correctly

ggggggggg, 2018-07-20:
I think we still have the record length of a `DataStreamProcessor` hardcoded somewhere, so that when I set the record length short (say 400 samples, 66 presamples), then quit and restart dastard, the record length is actually 256 samples. Though dastard commander says the right value, so clearly we had access to it.

ggggggggg, 2018-07-20:
I think the problem is related to this:
https://github.com/usnistgov/dastard/blob/45eb406206cc4b4694304f419fcce5a8dd0b4b2f/process_data.go#L74

ggggggggg, 2018-07-20:
It seems like ConfigureSource should include a NSamples and NPresamples argument.

joefowler, 2018-07-25:
Cannot reproduce with simulated pulses. Is this a Lancero-only thing?

Can you check your `~/.dastard/config.yaml` file after you quit dastard and before you restart it? Does it have the correct 400/66 value in its `status` object?

joefowler, 2018-07-25:
Very weird! I do see the problem.

commit 218aa17f (2018-08-16): Merge pull request #102 from usnistgov/restore_record_length

set pulse lengths after starting source, fixes #78
commit ef80758d (2018-08-16): set pulse lengths after starting source, fixes #78
closed 2018-08-16 by ggggggggg