usnistgov/dastard #85: Crash: PublishSync got stuck

joefowler, 2018-07-26:
After fixing #84 I deleted my `~/.dastard/config.yaml` file, so I was back to silly defaults. Try the following, and if it doesn't produce the same error, then try deleting your config file and repeating:

1. Set Simulated pulse source to 1 channel, 20 samples per second, 100 samples per pulse
2. Start Data

After a few seconds, dastard crashes thus:

```text
2018/07/26 09:45:12 Starting data source named SIMPULSESOURCE
2018/07/26 09:45:12 TriggerState: [{[0] {true 1s false true 4000 false true false 100 false false false 0 0 0 0}}]
2018/07/26 09:45:12 chanNames: [chan1]
2018/07/26 09:45:12 SEND Message of type dastard.ServerStatus: {"Running":true,"SourceName":"SimPulses","Nchannels":1,"Nsamples":4,"Npresamp":3,"Ncol":[],"Nrow":[],"ChannelsWithProjectors":[]}
2018/07/26 09:45:12 SEND Message of type []dastard.FullTriggerState: [{"ChannelIndicies":[0],"AutoTrigger":true,"AutoDelay":1000000000,"LevelTrigger":false,"LevelRising":true,"LevelLevel":4000,"EdgeTrigger":false,"EdgeRising":true,"EdgeFalling":false,"EdgeLevel":100,"EdgeMulti":false,"EdgeMultiMakeShortRecords":false,"EdgeMultiMakeContaminatedRecords":false,"EdgeMultiVerifyNMonotone":0}]
2018/07/26 09:45:12 SEND Message of type []string: ["chan1"]
2018/07/26 09:45:13 SEND Message of type dastard.Heartbeat: {"Running":true,"Time":0,"DataMB":0}
2018/07/26 09:45:15 SEND Message of type dastard.Heartbeat: {"Running":true,"Time":0,"DataMB":0}
2018/07/26 09:45:17 SEND Message of type dastard.Heartbeat: {"Running":true,"Time":0,"DataMB":0}
2018/07/26 09:45:17 PublishSync got stuck
```

Yeah, this is generating 5 seconds of data at a time and then feeding it forward. That's a pathologically long time. So maybe this is not an issue to be fixed. Thoughts?

ggggggggg, 2018-07-26:
I think you're analysis is correct, `PublishSync` has a timeout of 5 seconds so dastard doesn't just hang. Solutions.

1. I could increase it to 30 seconds, but you can still cause this problem by just making more ridiculous settings. 
2. I could make SimlulatedPulse source error if the settings are too ridiculous. 
3. I could try to calculate an update rate from the source, and pass that to publish sync, so it has a good timescale.
4. I could have publish sync learn the update rate based on observation.
5. do nothing

Of these I like 2 and 5.

joefowler, 2018-07-26:
I agree that 2 and 5 seem best. Maybe it's something to enforce in dastard-commander?

ggggggggg, 2018-07-26:
I would just put it in the `NewSimulatedPulseSource` function.

ggggggggg, 2018-07-26:
6. Just

commit 052cd681 (2018-08-15): fix #85
closed 2018-08-16 by ggggggggg