usnistgov/dastard #94: dastard writes framecount to the rowcount field in ljh22

ggggggggg, 2018-08-02:
Framecount only makes sense for a lancero source, `rowcount = framecount*nrow+row`

joefowler, 2018-08-13:
Did your comment above mean _rowcount_ only makes sense for a Lancero source?

Would you like to switch LJH2.2 to storing frame counts? Well, no switch is required, as that's what it already does. But I mean we'd change the text in the header AND update MASS to deal with it.

Or do you want to store row count, as with LJH 2.1?

ggggggggg, 2018-08-13:
Yes, I mean rowcount only makes sense for a lancero source. edited.

I can't decide easily if it should be framecount or rowcount in LJH2.1. I think I prefer rowcount because then less total code needs to change, and it will match the numbers for an external trigger if we ever get to that. But it does seem odd to write something that only makes sense for some sources. At least rowcount falls back to equal framecount when nrow = 1.

ggggggggg, 2018-08-16:
fixed by #102

closed 2018-08-16 by ggggggggg