usnistgov/dastard #104: dastard fails to keep up with lancero reads

ggggggggg, 2018-09-14:
Paul and I looked at this some... here are some observations
Shared settings: 133MB/s, 224 FB/error pairs, 10 ms auto trigger on all feedbacks (no errors), 150 record length, 75 pretrigger, 10 nsamp, mix = 0, projectors are loaded so real time analysis is happening
1. Running with Shared settings and nothing else is stable for 15 minutes at least (stopped after 15 mins with no errors)
2. Adding file writing (either LJH or LJH + OFF) causes instability, eg it crashes after approx 3 minutes
3. With file writing, setting nice = -20 does not prevent crash

Uniformly across our observations, the problem was preceded by processSegments taking 250ms to 900 ms, where normal is more like 1ms-50ms. Also, we instruments dsp.processSegment and see timings go from something typical like 
``` 
dsp ch 190, 1 6.089µs, 2 6.539µs, 3 25.353µs, 4 3.288931ms, 5 3.291853ms, 6 3.292221ms
```
to something like
```
dsp ch 408, 1 707ns, 2 1.221µs, 3 26.662µs, 4 22.97488ms, 5 22.977369ms, 6 22.977645ms
dsp ch 431, 1 161ns, 2 258ns, 3 4.087µs, 4 22.690602ms, 5 22.779968ms, 6 329.775695ms
```
Note that even channels share the longer 3->4 time (triggering), but only feedback channels have the longer 6 time (writing).

ggggggggg, 2018-09-14:
TODOs:
1. Show Joe
2. Ask Gene about linux scheduler
3. move lancero reading into it's own go-routine

joefowler, 2018-09-15:
Uh oh. Does this mean that we have made a huge mistake writing DASTARD in a garbage-collected language? How can we figure out whether GC is related to the problem?

Unless I'm mistaken, processSegments should not involve any I/O activities other than the writing. Also, you've identified that the problem happens only when writing. So it actually sounds like it's not a problem with GC but with writing.

All I can think of to investigate is whether our goroutines treat writing as a blocking activity. If so, can we replace the blocked goroutine with a send of the writeable data into a high-capacity go channel? Obviously we would still need the writer to keep up on average, but at least `processData` could proceed when writing gets temporarily behind.

I also asked you about whether we were making too many small writes, and you answered that we were using a Go library (or a system library?) for buffered writing. It occurs to me that buffered writes are a good idea, but they might still be inadequate for our purposes if the buffer size we choose (or the size we default to) is too small.

I don't know what size would be best, but I'll bet that the answer depends on the data-writing rate. That is, the best buffer size might be "enough to hold 0.2 seconds of data". It won't be easy to choose such a size dynamically. An alternative might be to write our own buffering object. Imagine a goroutine that listens to 3 channels. One notifies it of data records to write to disk at some later time; another might be a simple timer, waking it up every 0.1 or 0.3 seconds or whatever to write all accumulated records. The third, of course, tells it to quit. Ideally, the 224 such goroutines would not all wake up at once, either because they all start out of phase, or because we do something cute like assign each a different random sleep time between 0.1 and 0.3 seconds. I haven't looked at Dastard in a month, and I couldn't tell you how far this is from our current design.

ggggggggg, 2018-09-19:
Summary: We should probably refactor to do 1. all calls to WriteRecord then 2. all calls to Flush.

At the bottom is an MWE to explore writing performance in DASTARD.

You will notice that there are two potential locations to call `Flush`, each of which is gated based on a global variable at the top. The first location, currently enabled, is similar to how DASTARD currently works. Call flush after a block of records are written within a goroutine associated with a single channel. Typical ouptut with `flushWithinBlock = true`, `flushAfterBlocks = false`:
``` 
/var/folders/_0/25kp6h7x25v6vyjv2yjlcnkm000wrm/T/
recordsPerChanPerTick 10, Chans 500, tickDuration 50ms
records/second/chan 200, records/second total 100000
megabytes/second total 21.599999999999998
flushWithinBlocks true, flushAfterBlocks false
z, time.Now().Sub(tLast) 1 131.059939ms
writeSum 18.141417145s
flushSum 57.246504ms
z, time.Now().Sub(tLast) 2 384.410037ms
writeSum 1m36.424555762s
flushSum 32.473646ms
z, time.Now().Sub(tLast) 3 308.787389ms
writeSum 1m23.830675242s
flushSum 47.982118ms
z, time.Now().Sub(tLast) 4 317.128053ms
writeSum 1m6.803607974s
flushSum 52.000088ms
```

The second location, currently disabled, appears to work much better. Call flush after all blocks of records across all goroutines are written. Typical output `flushWithinBlock = false`, `flushAfterBlocks = true`:
```
recordsPerChanPerTick 10, Chans 500, tickDuration 50ms
records/second/chan 200, records/second total 100000
megabytes/second total 21.599999999999998
flushWithinBlocks false, flushAfterBlocks true
z, time.Now().Sub(tLast) 1 108.70203ms
writeSum 13.860676793s
flushSum 39.582µs
z, time.Now().Sub(tLast) 2 55.989804ms
writeSum 12.601853466s
flushSum 26.824µs
z, time.Now().Sub(tLast) 3 54.72021ms
writeSum 11.814723862s
flushSum 24.922µs
z, time.Now().Sub(tLast) 4 53.861773ms
writeSum 11.464808946s
flushSum 26.969µs
```

Finally, output with `flushWithinBlock = false`, `flushAfterBlocks = false`
```
recordsPerChanPerTick 10, Chans 500, tickDuration 50ms
records/second/chan 200, records/second total 100000
megabytes/second total 21.599999999999998
flushWithinBlocks false, flushAfterBlocks false
z, time.Now().Sub(tLast) 1 110.439406ms
writeSum 14.171515231s
flushSum 45.202µs
z, time.Now().Sub(tLast) 2 52.745028ms
writeSum 10.485814479s
flushSum 25.853µs
z, time.Now().Sub(tLast) 3 50.947804ms
writeSum 10.981908239s
flushSum 25.338µs
z, time.Now().Sub(tLast) 4 51.520545ms
writeSum 12.228746899s
flushSum 25.461µs
```

**Apparently interspersing calls to write and calls to flush leads to terrible results.**

```
package main

import (
	"fmt"
	"os"
	"os/signal"
	"sync"
	"time"

	"github.com/usnistgov/dastard/ljh"
)

const flushWithinBlock = true
const flushAfterBlocks = false

func main() {
	dirname := os.TempDir()
	fmt.Println(dirname)
	recordLength := 100
	N := 500
	recordsPerChanPerTick := 10
	writers := make([]ljh.Writer, N)
	abortChan := make(chan struct{})
	for i := range writers {
		writers[i] = ljh.Writer{FileName: fmt.Sprintf("%v%v.ljh", dirname, i), ChannelIndex: i, Samples: recordLength}
	}
	go func() {
		signalChan := make(chan os.Signal)
		signal.Notify(signalChan, os.Interrupt)
		<-signalChan
		close(abortChan)
	}()

	tickDuration := 50 * time.Millisecond
	ticker := time.NewTicker(tickDuration)
	z := 0
	tLast := time.Now()
	fmt.Printf("recordsPerChanPerTick %v, Chans %v, tickDuration %v\n", recordsPerChanPerTick, N, tickDuration)
	fmt.Printf("records/second/chan %v, records/second total %v\n", float64(recordsPerChanPerTick)/tickDuration.Seconds(), float64(recordsPerChanPerTick*N)/tickDuration.Seconds())
	fmt.Printf("megabytes/second total %v\n", float64(recordLength*2+8+8)*float64(recordsPerChanPerTick*N)/tickDuration.Seconds()*1e-6)
	fmt.Println("starting core loop")
	for {
		z++
		select {
		case <-abortChan:
			fmt.Println("clean exit")
			return
		case <-ticker.C:
			var wg sync.WaitGroup
			writeDurations := make([]time.Duration, N)
			flushDurations := make([]time.Duration, N)
			for i, w := range writers {
				records := make([]Record, recordsPerChanPerTick)
				for j := range records {
					records[j] = Record{framecount: int64(z*10000 + j), timestamp: int64(z*10000 + i), data: make([]uint16, recordLength)}
				}
				wg.Add(1)
				go func(w ljh.Writer, records []Record, i int) {
					tStart := time.Now()
					defer wg.Done()
					for _, record := range records {
						if !w.HeaderWritten {
							err := w.CreateFile()
							if err != nil {
								panic(fmt.Sprintf("failed create file: %v\n", err))
							}
							w.WriteHeader(time.Now())
						}
						w.WriteRecord(record.framecount, record.timestamp, record.data)
					}
					tWrite := time.Now()
					if flushWithinBlock {
						w.Flush() // Flush here for terrible performance
					}
					writeDurations[i] = tWrite.Sub(tStart)
					flushDurations[i] = time.Now().Sub(tWrite)
				}(w, records, i)
			}
			wg.Wait()
			for _, w := range writers {
				if flushAfterBlocks {
					w.Flush() // Flush here or not at all for reasonable performance
				}
			}
			fmt.Println("z, time.Now().Sub(tLast)", z, time.Now().Sub(tLast))
			var writeSum time.Duration
			var flushSum time.Duration
			for i := range writeDurations {
				writeSum += writeDurations[i]
				flushSum += flushDurations[i]
			}
			fmt.Println("writeSum", writeSum)
			fmt.Println("flushSum", flushSum)
			tLast = time.Now()
		}
	}

}

// Record does stuff
type Record struct {
	framecount int64
	timestamp  int64
	data       []uint16
}
```

ggggggggg, 2018-09-19:
I tried a lot of different things in Dastard to get this problem to go away...

1. Remove all calls to `Flush` (eg rely only on 32kB buffer in each file), still saw long pauses.
2. Call `Flush` as `flushAfterBlocks` above, still saw long pauses.
3. Call `Flush` on 1/20 channels on each 50 ms iteration, with each channel hit once/second. Still saw long pauses.
4. Write to the SSD drive in the RAVEN pc (not sure exactly what software config). Still saw long pauses.
5. Move all file writing to occur in a single goroutine sequentially (with the 1/20th flush strategy following) after all processing. Still saw long pauses.

Also when I do have `Flush` calls, the pauses seem to come from the flush calls.

Oddly, despite result #2, when I run the above testing code on the RAVEN pc with `flushAfterBlocks` enabled, I don't see any anomalous pauses. 

So I think my conclusion is that somehow DASTARD is working worse than my test program, but I don't know why.

ggggggggg, 2018-09-28:
#105 fixes this by adding a goroutine which only reads from the lancero and demuxes and puts data into a channel. the channel acts as a buffer allowing up to 5 seconds of lag in data handling.

joefowler, 2018-09-28:
Fixed in #105

closed 2018-09-28 by joefowler