usnistgov/dastard #106: Simulated pulse source fails with a map?

ggggggggg, 2018-09-28:
```
Traceback (most recent call last):
  File "dc.py", line 145, in updateReceived
    self.observeTab.handleTriggerRateMessage(d)
  File "/home/pcuser/dastard-commander/observe.py", line 46, in handleTriggerRateMessage
    self.buildCRM()
  File "/home/pcuser/dastard-commander/observe.py", line 95, in buildCRM
    self.crm_grid = CountRateMap(self, self.cols, self.rows, self.getChannelNames())
  File "/home/pcuser/dastard-commander/observe.py", line 90, in getChannelNames
    assert(len(channel_names) == cols*rows*(1+self.auxPerChan))
AssertionError
Aborted (core dumped)
```

joefowler, 2018-09-28:
Isn't this a dastard-commander issue?

ggggggggg, 2018-09-28:
Umm, yes. I'll move it.

closed 2018-09-28 by ggggggggg