usnistgov/dastard #107: Make simulated pulse source have more than 1 line

ggggggggg, 2018-09-28:
To be able to test end to end data analysis with calibration routines it would be nice to be able to set a set of pulse heights for simlated pulse source, instead of just 1.  Could just cycle through them for now.

joefowler, 2018-10-05:
That makes sense. But I'd hate to add more and more configuration info to the GUI.

Suppose instead of making a single pulse of the requested size, we repeat a sequence of pulses that are size [0.6, 0.8, and 1.0] times the requested size? That would be the easiest. Would that meet your calibration needs?

ggggggggg, 2018-10-05:
I suspect that would be fine. We could maybe add an API to change the sequence, but hardcode the sequence in dastard commander?

PR #112 (2018-10-05): Simpulse accepts/requires a slice of amplitudes
Will require the new version of dastard-commander to talk to it.

Fixes #107
commit 8af7a7de (2018-10-05): SimPulse source requires a list of amplitudes

This lets us make pulses of more than 1 size. Fixes #107, though we'll also need to change DC.
closed 2018-10-05 by joefowler