usnistgov/dastard #114: Decide how to publish fb/err and row x column info

joefowler, 2018-10-08:
See [Microscope issue 19](https://github.com/usnistgov/microscope/issues/19) for related discussion. How do we get microscope to present sensible choices to the operators? I think the answer is that the pulse record header needs to include at least:
* A data source code (so that TDM systems can be assumed to have err/fb/err/fb channel pattern).
* A number of data channels per "group" in the data array (=TDM columns, or uMUX channels per sub-band). Here a "group" means all the data that are read out in order before the next group, so a single row of TDM data, or a channel count in a sub-band of uMUX.
* A number of "groups" in the data array (=TDM rows, or uMUX number of sub-bands).
* A channel name for this channel, e.g. "chan 1" or "err 0", which can be a variable-length string or fixed-length.

Or instead of making this a record header, it could be a second sort of message on the pulse records PUB socket. Assume that microscope cannot plot until it has received this kind of message, so one would precede every actual pulse record message (except if one has been sent in the last 1-2 seconds?)

Or perhaps the second sort of message can be sent every 10 seconds OR when requested by client. This way, dastard-commander can make such a request when user clicks "launch Microscope". Users who start Microscope at the command line will have to be more patient, perhaps (or we can make Microscope be a temporary RPC client?)

The total data here is probably only 13 bytes: 8 for the var-length channel name, and 1 each for the name length, data source code, and number of groups; to be safe, use uint16 for the number of channels per group, in case uMUX ever needs > 256 channels per sub-band? 13 bytes isn't a lot, so it would be cheap to make this a second frame in every ZMQ message. The problem? That microscope has to decide which channels to subscribe to before it learns the "array geometry" info.

Thus my proposal has two versions. The complicated one:
1. Generate a new kind of message to send on (base port)+2 = 5502. It will have a negative magic number that microscope can always subscribe to. It will contain the array data source code and rows x columns. This message is sent every 10 (?) seconds by default.
2. Add an RPC command `PublishArrayGeometry` to dastard. It will generate the relevant message. Optionally, it can also send some message on the status port (5501).
3. Dastard-commander will need to make that RPC request whenever user clicks *start Microscope*. (This will have a race problem: no way to know when the newly started microscope instance is ready to process such a message.)
4. Microscope will have to process this message to fix its own issue 19.

I think the above is needlessly complicated, now that I've written it out. Instead, I favor the simpler one:
1. Generate a new kind of message to send on (base port)+2 = 5502. It will have a negative magic number that microscope can always subscribe to. It will contain the array data source code and rows x columns. This message is sent every 1.5 (?) seconds by default.
2. Microscope will have to process this message to fix its own issue 19.

joefowler, 2018-10-08:
@ggggggggg can you review this? Once we agree, we can close this, and I'll start one or more issues for the actual work.

ggggggggg, 2018-10-09:
This message could serve a dual role, it could signal "all future pulses will have a framecount greater than or equal to this number" and you could send it out in every iteration of `CoreLoop`. Negative magic number sounds good. I don't super like the idea of having a 1.5 second timer as it adds another timer, it will make microscope slow to startup, and seems a bit arbitrary. I'm also fine with just putting it in each pulse record.

joefowler, 2018-10-09:
What would that second role of each message help us to do? I don't get it. 20x per second sounds pretty excessive, but if we can think of why microscope (or other subscribers) could use this message, then we can do it. Obviously, the number of pulse messages at 200 cps x 240 detectors is 50k per second, beside which 20 per second is practically nothing.

On further reflection, I'm starting to feel strongly that we need the new info to be a separate message, because of ZMQ packet filtering. We want microscope to ignore the pulse records for most channels, to reduce total traffic, both on the network and into microscope. That means we need to not subscribe to messages from 95% of all channels. If we do that, though, then how will we ever receive a message in order to parse its array geometry information? We can arbitrarily pick a few channels but we'll have to hope they are not all dead. We can pick all channels, but then we get flooded with messages.

Better to subscribe to messages with the special array geometry info magic number, then process one such message, then build the GUI elements, then start subscribing to the desired channels.

joefowler, 2018-10-09:
Wait a minute, here's a simple idea: what if we just require microscope to be *started* with the relevant information. All we need is a flag to select among TDM/uMUX/etc, a number of rows, and a number of columns. That could all go on the command line, and dastard-commander could be tasked with setting the flags correctly. The first flag, in fact, could even just be a true/false flag for whether this is TDM (e.g. use `-n` or `--no-error-channel` for a non-TDM system).

This would make a great many things simpler: Dastard can be unchanged; DC just needs to launch the microscope command with certain flags; and most important, microscope can handle the information exactly the way we want: once at startup.

ggggggggg, 2018-10-09:
Starting microscope with command line info sounds good. Nice and simple. I guess it just crashes if anything changes? Not elegant, but fine (also I think matter fails to handle changes sometimes, so probably more likely to work).

joefowler, 2018-10-09:
Hmm. Good question: what if something changes? I don't think anything would cause a crash:

* If you change TDM to SimPulses (i.e. a source with error/FB to non-error source), then half the channels will get interpreted incorrectly. Your loss, but it's not a crash.
* If you increase the number of channels, then the new higher ones will be unreachable from microscope. (Or maybe not, if you type in the various boxes?)
* If you decrease the number of channels, then the higher ones that have gone away just won't ever send messages.
* If you change the number of rows and/or columns, then the drop-down menus won't make sense for the new geometry, but again not a crash.

joefowler, 2018-10-09:
Okay, let's go with the mods to microscope only. If you had a reason for sending out a promise about what the future minimum frame number will be, then we can discuss. I still didn't see what that really helps.

closed 2018-10-09 by joefowler