usnistgov/dastard #125: Read Abaco devices

joefowler, 2019-01-25:
The Abaco hardware for the Xilinx FPGA µMUX system needs to be readable by Dastard as a new implementation of the `DataSource` interface. It should present as a device-special file that can be read in chunks of roughly MB sizes (the data will be transferred from the Abaco to memory by DMA).

Several questions to be answered:

- [ ] 1. What happens to the device `c2h` ("card to host") DMA data when we aren't reading it? Does it fill some buffer? Is that an error? Will a full buffer contain old, rather than new, data, so that we always have to discard 1 buffer worth of data to start?
- [ ] 1. Are there definitely going to be 16 bits per word? 32? If variable, how do we pick? If 32, can I just chop it down somehow to 16? Dastard will be much easier if it can be 16-bit data.
- [ ] 1. What does the Dastard need to know about the Abaco, and how will it know it? We can infer the # of channels as with Lancero, but we'd much rather just know.  Also seems like we ought to be told the nominal data rate and bits per word (if this varies). Wouldn't be bad to also know some kind of hardware ID/name, and maybe a version number for the data format. Can we create something on the /sys/ filesystem to share info like this?
- [ ] 1. What flexibility do we need for the future? Do we need to consider the possibility of 2+ Abaco cards per PC, or of multiple parallel `c2h` devices on one card?

My hunch is that this will be an easier implementation than Lancero, but we will need to answer the above questions.

Including @DarkTyr and @ggggggggg.

joefowler, 2019-01-25:
See branch abaco in this repo.

ggggggggg, 2019-01-28:
Also, how long do we have between reads before we lose data? I thought it was 200 microseconds, but have seen comments suggesting that it is longer.

joefowler, 2019-01-28:
I've been reading John's code (C++ and Python) that do simple reads of the device. It appears I had forgotten an important part: he is now using Xilinx libraries to make DMA happen (Direct Memory Access). There are subtle parts that we'll need to ask him about, but the bottom line is that user-space code can read data from a buffer managed elsewhere (Xilinx driver??), a buffer that is much larger than a single DMA transfer or the FIFO size inside the Xilinx firmware. (It's this latter object that needs to be emptied every 200 µs). 

Let's picture this buffer being something like 1 MB or more. Dastard only needs to ensure that this buffer never fills when we are actively reading data from the Abaco. I believe our top rate is around 256 MB/sec, so that means we have at least 4 ms to read, or 16 ms if we make the buffer 4 MB.

ggggggggg, 2019-01-29:
I see, 16 ms is much much more reasonable. Can we get it up to 50 ms? in that case it's the same rate as the lancero currently is, which seems like it might be useful.

joefowler, 2019-01-29:
Well, we'll need to speak with John to see what the issues are. The problem I foresee is that "50 ms" is not a buffer size. You have to know the data rate to convert it to a buffer size. 

We can assume a certain data rate and then choose the buffer size to follow, but if we later do a test with a much lower data rate, then the buffer takes forever to fill. That's okay, except for the problem that the first 1 full buffer worth of data has to be drained before you can read data and be sure it isn't stale data.

One possibility: maybe the device file can answer the query "how many bytes can I read at this moment without blocking?" I think that's a pretty standard thing. If so, then we know how many bytes have to be discarded. Normally the answer will be 1 full buffer, but if data is being acquired slowly and the buffer was recently read, then perhaps it can be less. I'll add this to the above list of discussion topics to ask John about.

ggggggggg, 2019-01-29:
Good point, I forgot about the data rate issue. With the lancero currently we have the buffer size set such that at max data rate it takes more than 50 ms to fill even halfway, then we try to read from it every 50 ms. 16 ms is probably fine, but it would be useful to know what the maximum size available to us is, and what time it corresponds to at max rate.

joefowler, 2019-01-29:
An alternative to the "how many bytes are available" question is the classic "flush thyself" command. 

Somewhere in between is the non-blocking read. Maybe the driver already supports this: where you say "read up to 4 MB but return immediately", and it says "here's 1 MB, which is all I have." Our strategy will depend on what the Xlinix DMA driver supports.

Here's my ideal:
* The Abaco setup code ensures that the DMA buffer is at least big enough to hold 100 ms of data at the current rate or--if rates are variable--at the maximum anticipated rate (we could probably negotiate this down, but as you say, 100 ms is the Lancero figure). This is done before Dastard starts.
* When Dastard starts, it somehow learns the DMA buffer size and the anticipated time that would fill that buffer.
* When Dastard goes into Abaco data acquisition mode, it creates its own `[]byte` buffer of the same size.
* It immediately does a non-blocking read of (up to) the full buffer size and ignores the data that it gets. This is equivalent to a flush of all stale data.
* Now it can start a `Ticker` to awaken every (say) 50 ms that reads all available data.

joefowler, 2019-01-29:
John's [README for xdma_software](https://bitbucket.org/nist_microcal/xdma_software/src/master/)  points to [Xilinx Answer Record 65444](
https://www.xilinx.com/support/answers/65444.htm) about the driver.  ln turn, it links to a [software guide PDF](
https://www.xilinx.com/Attachment/Xilinx_Answer_65444_Linux.pdf) for their PCI express DMA drivers. I will read this.

joefowler, 2019-01-31:
I've discussed the details with John, and I think I have cleared up enough of my misconceptions that it makes sense to close this issue and replace it with #126.

closed 2019-01-31 by joefowler