usnistgov/dastard #126: Read Abaco devices (revised version)

joefowler, 2019-01-31:
The Abaco hardware for the Xilinx FPGA µMUX system needs to be readable by Dastard as a new implementation of the DataSource interface. It should present as a device-special file that can be read in chunks of roughly MB sizes (the data will be transferred from the Abaco to memory by DMA).

This issue replaces #125, which contained a lot of misunderstandings in my questions and comments. We'll be better served by staring a new issue.

The Xilinx driver with DMA offers two key devices:
1. `/dev/xdma0_c2h_0` : a "card-to-host" DMA device on card number 0, channel number 0. We open this file for reading, then we make system `read(...)` calls to it. I think there can in principle be additional cards and additional channels per card.
1. `/dev/xdma0_user` : an interface for user information. I don't know the details yet, but we can apparently use mmap and/or reads from this device to find out key facts like the data rate, size of the firmware FIFO, firmware version number, etc. It memory maps computer memory to an 8 kB RAM block in the firmware.

**Reading the c2h device**: The system read calls require us to pass in an allocated user-space buffer to be filled and a buffer size. Besides this, there is no software buffer. If we have not been reading data recently, then the FIFO will be full of arbitrarily old ("stale") data. For reasons John cannot work out, it is necessary to read 1 FIFO (currently, 64 kB) plus epsilon (~2 kB) to get to non-stale data. Perhaps safest just to read 2 FIFOs worth and discard them.

To be clear: the DMA does not start until a `read()` call requests it. Then we block on the read. (This is a reason it's good to know the data rate from the user device: so we can be sure the read size is matched well to the data rate and the desired time required to read.) When the blocking read returns, we will need to do two things concurrently: start the next blocking read, and process the results of the one that just completed. Go is a good language for this kind of thing.

On the data word size: currently John is planning 32 bits, including 2 non-data bits (both are analogous to the "frame bit" and the sync bit from TMD). He seemed open to switching to a header+data format in which the info communicated by the frame bit and the sync bit go into a header block. This would let us switch to 16-bit data without having the actual data have to pack into 14 bits.

Again, there is no buffer in driver space, being filled continuously, from which we must copy. That's the Lancero / TDM model. Instead, the Xilinx DMa driver lets us provide user-space buffers, which it will fill upon request. The Abaco model in Dastard might look like this:

1. In Dastard's "sample phase", query the "user device" to learn the data rate, data format (if that becomes a variable), and how much initial data will be stale.
1. Still in sampling phase, read enough data from the Abaco to discard the stale data and analyze what follows to determine channel and row counts. 
1. Figure out how many bytes not yet read will constitute the end of the current frame. Read those bytes, so that the next read will be at the start of a data frame. Now switch to run phase.
1. In run phase, we create a data buffer of type `[]byte` and read into it (blocking). When the read is done, we launch a goroutine to process the buffer and--if not ordered to quit--immediately repeat the process of creating a buffer and reading into it. We might want the read to be done in a goroutine and the result sent back on a channel, so that we can easily cancel during a read either because a timeout expires or the user quits.

joefowler, 2019-01-31:
_Useful driver info copied from #125:_ John's [README for xdma_software](https://bitbucket.org/nist_microcal/xdma_software/src/master/)  points to [Xilinx Answer Record 65444](
https://www.xilinx.com/support/answers/65444.htm) about the driver.  ln turn, it links to a [software guide PDF](
https://www.xilinx.com/Attachment/Xilinx_Answer_65444_Linux.pdf) for their PCI express DMA drivers. I will read this.

The driver offers two modes for DMA transfers: interrupt-driven and polling. John says that the data rates we anticipate will require us to use the polling mode. Not sure what the consequences of this are, but if that's so, then this is one decision taken out of our hands.

ggggggggg, 2019-01-31:
So in this picture, assuming 64kB corresponds to 200 us, we have 200 us to launch a goroutine to process our buffer, and call read again. That sounds do-able.

joefowler, 2019-01-31:
Right. That's the timescale. I have no idea how we can guarantee never to miss the start-read window. Probably we can't. I'll bet it can be made to work, though. Perhaps Dastard can be robust against missing some frames? I mean, the data would be mangled, but maybe we can keep the server itself from crashing?

joefowler, 2019-01-31:
> There may be a little more behind the scenes that needs to be done, but it depends on how/what the Go file read actually does. C vs C++ vs Python each are actually a bit different and have to be treated differently. For instance:
> 
> a) C++ you can just call a read for all of your data (has to be a set amount) and then you have to fully close the file. The close and open process takes a lot longer than one would think. Reference this [commit](https://bitbucket.org/nist_microcal/xdma_software/src/master/C/xdma_acquire/xdma_acquire.cpp) on the master branch of xdma_software.
> 
> b) the C implementation takes more work but can just be looped over on the read (more applicable to DASDARD) but you have to keep track of the return value and the number of bytes read. Even in polling mode it can return with less than requested but not all the time. When the driver is in interrupt mode and the read is truly non-blocking I felt it got a bit wonky in how it worked. You still had to loop over the read. [This Commit on master](https://bitbucket.org/nist_microcal/xdma_software/commits/f4d541edbe2d741669507e13cc56fb4e77f6ce4b#chg-C/xdma_testing_scripts/xdma_acquire.cpp) look at xdma_acquire.cpp
> 
> c) Python you could read some data, then go back later and read more data without issues, but looping over the read was no where near fast enough. It seemed like there was a lot of setup time before performing the hardware read. To clarify, this is using the read built-in to Python, not calling my C++ program. [Python Read](https://bitbucket.org/nist_microcal/xdma_software/src/master/python/nist_4dsp_lib/xdma_stream.py) Method xdma_read. xtake_data calls the C++ program and uses the stdout pipe to transfer data.
> 
> Before we think to much harder on this, we have to figure out how Go reads the device file. That will inform the rest.
> 
> Another question you had posted was support for multiple cards, the answer is most likely. The driver automatically handles this and just changes the xdma number, xdma0_user and xdma1_user with two cards in the system.

And similarly, can there be two channels per card, e.g. xdma0_c2h_0 and xdma0_c2h_1?

My plan for now is to support _N_ channels per card, maybe 8, and a single card. But I'll try to make it easy to add a second card later. What could potentially screw that up is the possibility of the two cards acquiring at different rates. (Same problem we have with _N_>1 TDM cards.)

joefowler, 2019-02-01:
John, I was curious about your claim that reading 1 MB in C++ returns 1 MB, but doing the same in C returns 16 kB. I couldn't find any pure C programs in the repo, so I wrote a new program `Cacquire.c` in C (obviously). As I suspected, the difference is not the language, but whether you're doing buffered or unbuffered reads.

**Unbuffered, low-level reads**: these use functions `open() read() close()` in the C library, each of which is a system call (or a  simple wrapper around one). When I call `read(file_descriptor, buffer, 1024*1024)` several times, they return only 16384 bytes (except a different number the first time).

**Buffered, high-level reads**: these use the C library functions `fopen() fread() fclose()`, which are the buffered versions. They can hide an unknown number of system calls behind the veneer of one `fread()` call. When I use these, I get the full 1 MB (again, with a slightly different figure the first time).

In golang, I was using the unbuffered, low-level file reading. Presumably I could get the 1 MB reads that we want by switching to its buffered IO mechanism. But I want to pause first and consider whether that's the right approach. 

I wish I understood _why_ the low-level read calls never return more than 16 kB. It feels like it would be more efficient for our needs if the reads would pause (block) until the desired amount of data is available. Is it possible that running the driver in interrupt mode would solve this problem? Perhaps polling mode means non-blocking reads? You claimed that interrupt mode would be inefficient at high rates, but perhaps that's only true if you try to make small reads? We'll discuss in person....

DarkTyr, 2019-02-01:
The commit link I provided above for C is not pure C but used C constructs for the file read (copied from Xilinx). Thus as you found out, it behaves the same. It doesn't have to be pure C to show that. So that seems like a waste of time. Buffered versus Unbuffered, I guess that shows my lack of knowledge in programming, which might be why I have been asking for help for a long time. 

If it is determined to be a problem later on then we can revisit this buffered versus unbuffered read method. For the buffered read call in C, can you go and make a second read after the first without closing and opening the file? Because if you can't this is pretty much a done deal here. 

Do you want to switch driver operation (rmod and insmod) the driver? I don't. You can try interrupt mode on the driver, but that is not how we will need to operate. Just run ./load_drive.sh instead of the ./load_driver_polled.sh if you would like to try.

joefowler, 2019-02-01:
I did try switching the driver into interrupt mode. It failed. In short, the low-level calls still returned only 16 kB, while the high level calls would return data only once (calls after the first returned 0 bytes of new data). I don't understand this, but I know it's the opposite of progress, and so I returned the driver to polling mode.

To answer your question: yes. In C, whether buffered or unbuffered, reads could be done 8 times in a row and always get data without closing the file. (Well, that's true only in polling mode. When the driver was in interrupt mode only the first read would return data, as mentioned above. Is there any chance that _your_ previous troubles with having to close then re-open the device file occurred in interrupt mode?)

Here's my plan: write the Golang acquire program to use _buffered_ IO and see what happens. If it appears to work well, then next we'll want to check carefully for dropped data.

joefowler, 2019-02-01:
Buffered IO in golang (`buffo.Reader`) did not solve the problem. It's designed to let multiple small reads from go be unpacked from a single underlying system call. We wanted the opposite: small reads from multiple system calls to be bundled into one Go read. (And even if it did what we want, it would surely be subject to interruption by the garbage collector, so I don't think this was a promising path in the first place.)

Looks like we need a C program to run a tight loop (no garbage collector!) filling a large C buffer (1-10 MB) and somehow shipping the packaged data to Dastard. 

Galen and I like the idea of using a Posix FIFO (a.k.a. "named pipe"), which you create with shell command or C library function `mkfifo`. It's a persistent memory-resident pipe (it has a place in the filesystem but otherwise has no relationship with the drive that it lives on). 

One problem for us is that you can't open a pipe for writing unless it has one or more readers. What we'd _like_ to do is open the pipe in advance (when the driver is loaded or when the Abaco is configured), then have writes go down the /dev/null drain until Dastard connects and reads from the pipe. Pipes don't have those semantics, however.

One approach suggested on stack exchange is to have the C program open the pipe for write _and read_. Then you don't actually read from that end. This behavior is apparently undefined by Posix but works on Linux. This has the problem that the pipe will quickly fill up with unread data. I suppose you could make sure to do a non-blocking read of 1 MB before every write of 1 MB of new data, to keep the pipe "cleaned out".

Or a different approach. Perhaps our C-language writer can have two loops. Which loop it runs in depends on whether a process is consuming Abaco data or not:
1. The first loop tries to open the pipe for writing. It uses a _non-blocking_ call to `open`, so it can return immediately to signal that the pipe has no reader. If no reader, repeat in 100 ms (?). If the open succeeds, move to the next loop. Notice that there's no need to read from the Abaco device during this loop, because no one is listening for its data anyway.
2. Now there's a pipe open for writing. Loop over reading the XDMA, building up a 1-10 MB block of data out of 16 kB chunks. When the larger block is complete (or after some timeout like 10-100 ms, whichever is first), write the entire block on the open pipe. If this fails (say, because the reader disconnected), then discard our XDMA data block, close the pipe, and return to the first loop.

I don't know how hard this will be. What if we're in loop 2 because the pipe is open for reading by Dastard, but Dastard does not actually consume the data? Can we preemptively give up before the pipe uses up all of system memory?

joefowler, 2019-02-06:
Update: I have written a simple data extraction daemon `deed` that puts data onto a named pipe. It has the twin loops mentioned in the comment before last: first to open the pipe for writing (retrying until a reader exists), then to read the device and copy to the pipe. It occurs to me that we can replace the first loop with a blocking open (the default) and block as long as needed. We actually don't care if the pipe can't be opened for writing until it's opened for reading. That's fine with us.

`deed` requires about 100% of a CPU if the XDMA driver is in polling mode, and about 400% if it's in interrupt mode. The latter is spread among 4 processes:
* 140% `rsyslogd`
* 100% `systemd-journal`
* 100% `deed`
* 60% `kworker`
It's clear, then, that polling mode is preferred. We can later investigate various modes of the driver to see if the 100% load can be reduced. John expects so, but we have very little to go on aside from random people's online postings!

Also, I have the answer to the above question:

> What if we're in loop 2 because the pipe is open for reading by Dastard, but Dastard does not actually consume the data? Can we preemptively give up before the pipe uses up all of system memory?

It turns out that the pipe writer will block once the pipe buffer is full. We can raise the pipe buffer size from its default (64 k?) in recent versions of Linux, but only as high as 1 MB. Once the pipe is that full, it will not accept writes. The `deed` will block on writing to it and will not keep consuming Xilinx data. That's a pretty safe thing.

Next step: try reading this data pipe into Dastard.

joefowler, 2019-04-12:
Two months have passed, and we've learned a great deal about this problem. Following Maszi's suggestion, we are planning now to replace the named pipe (FIFO) with Posix shared memory. The FIFO was limited to 1 MB of buffering, and it required an extra copy into and out of kernel memory. Shared memory can be unlimited in size and is not in kernel space.

I have implemented a lock-free ring buffer in both C and Go. The latter is an object `RingBuffer` that is carefully tested. The plan: write to that shared memory buffer from `deed`, and read from it in Dastard. Once it fills, new data will have to "fall on the floor", because a lock-free ring does not permit the data producer to clear out old data. (If it did, then the producer would in effect become a second consumer, and this design does not define behavior if consumers are contending for data.)

Let the shared memory regions use the same names as the /dev files that they are mirroring, e.g. `xdma0_c2h_0`. There can be one per /dev file.

Next steps:
1. Start filling that buffer with Abaco data.
1. Adjust the `AbacoDevice` and `AbacoSource` object in Dastard to use them.
1. Test on a real system.

joefowler, 2019-04-29:
There remain problems, but the core requirement of reading from Abaco is working.

closed 2019-04-29 by joefowler

ggggggggg, 2019-04-29:
So are we ready to test on a real system?

joefowler, 2019-04-29:
No. We are not ready. Wait until you see the list of specific issues that I am about to generate... Two so far that I know of, but that might be only the tip of the bugberg.

commit 7c820e69 (2019-04-29): Fix hang in Abaco read loop

Fixes #126.