usnistgov/dastard #127: Read Roach roach devices

ggggggggg, 2019-02-13:
We need to add support for roach devices. @joefowler 

Here are links to the implementation in matter: 
 * https://bitbucket.org/nist_microcal/nasa_daq/src/master/ndfb_server/RoachUDP.cpp
 * https://bitbucket.org/nist_microcal/nasa_daq/src/master/ndfb_server/RoachUDPProcessor.cpp

Do we have some software that generates Roach likes UDP packets, or any other documentation that would be useful? I'll email this issue to relevant people not on github.

DarkTyr, 2019-02-13:
Joe or Dan has a program that creates ROACH Droppings. AKA simulated data.

joefowler, 2019-02-13:
The program John is referring to is called PEST.  See [nasa_daq repository](https://bitbucket.org/nist_microcal/nasa_daq/src/master/pest/). I don't see a README, but the ROACH droppings are found in [roachpackets.cpp](https://bitbucket.org/nist_microcal/nasa_daq/src/master/pest/roachpackets.cpp).

ggggggggg, 2019-02-13:
I was able to get PEST to run by doing 

```
qmake
make
./pest
```
So that looks good.

ggggggggg, 2019-02-13:
Here is a spreadsheet with a written description of the data format:

[ROACH2 Packet Format.xlsx](https://github.com/usnistgov/dastard/files/2862095/ROACH2.Packet.Format.xlsx)

joefowler, 2019-03-25:
This is fixed in 6b8dc02, which we'll merge soon. I ran it on 16-channel PEST data for 20 hours without a single dropped or mis-ordered UDP packet. First, I want to fix an unrelated problem that causes Travis tests to fail.

ggggggggg, 2019-04-16:
The current roach firmware sends 4 word data, dastard assumes 2 word data. As a temporary fix I used this funciton 
```
func parsePacket(packet []byte) (packetHeader, []RawType) {
	var header packetHeader
	buf := bytes.NewReader(packet)
	if err := binary.Read(buf, binary.BigEndian, &header); err != nil {
		fmt.Println("binary.Read failed:", err)
	}
	wordLen := int(1 << (header.Flags & 0x3)) // data word length in bytes
	if wordLen == 4 {
		data := make([]RawType, header.Nchan*header.Nsamp*2)
		if err := binary.Read(buf, binary.BigEndian, &data); err != nil {
			fmt.Println("binary.Read failed:", err)
		}
		data2 := make([]RawType, header.Nchan*header.Nsamp)
		for i,_ := range data2 {
			data2[i] = data[i*2]
		}
		return header, data2
	} 
	if wordLen == 2 {
		data := make([]RawType, header.Nchan*header.Nsamp)
		if err := binary.Read(buf, binary.BigEndian, &data); err != nil {
			fmt.Println("binary.Read failed:", err)
		}	
		return header, data		
	}
	panic("shouldn't reach here")
	return header, nil
}
```
And now the microscope plots look nice. But the data falls into the range 0-65k, but it should be like -32k to +32k. I suspect this is actually microscope's responsibility? It looks like dastard is sending the raw data plus a `dataType` value that depends on if the data is signed or not.

ggggggggg, 2019-04-17:
Microscope is supposed to interpret the data as signed using the code here:
https://github.com/usnistgov/microscope/blob/4a4f05bebe02f4c3110ceddc64108431b9ecdd8a/pulserecord.cpp#L17

We should debug if microscope is being told it is signed or not.

PR #139 (2019-04-19): Roach data source is beginning to work
We have a basically working version of Dastard that will handle ROACH packets. Fixes #127 .
closed 2019-04-19 by ggggggggg