usnistgov/dastard #128: Remove projectors and basis from off file headers

ggggggggg, 2019-02-15:
Currently, off file headers includes the projectors and basis used encoded as base64. I wanted to make sure we always had access to this information, so putting it in the header seemed reasonable. But, it makes the headers take a longer time to parse (10s of seconds for 164 files). Also it makes the "human readable" JSON header very hard to read because the vast majority of text is just encoded projectors and basis. 

**Proposal 1**
Write a separate "off supplement" file that just contains the projectors and basis for every channel. It could be in JSON with parsing delayed until actually needed, or it could be a json header followed by binary data.

**Proposal 2**
Write the projectors and basis as binary data following the human readable header, proceeding the binary data for the records. Then we can mmap the projectors and basis. 
Currently we have
```
JSON header
newline
records (binary)
```
Instead it could be
```
JSON header
newline
projectors (binary)
basis (binary)
records (binary)
```
In this way, when parsing the header, you would simply mmap the projectors and basis, which takes almost no time.

I like proposal 2.

joefowler, 2019-02-15:
Yes. I agree with proposal 2, also. It’s kind of a hack to have a text-newline-binary format. However, it is no less of a hack for that binary section to contain a single data array than three different kinds of arrays, right?

ggggggggg, 2019-02-15:
I don't really understand your statement, but I think not that ugly to have 3 different arrays following the header. At first it seemed odd, but now that I've thought about it more, it seems pretty straightforward. It just requires a bit more math to figure out where each array starts.

joefowler, 2019-02-15:
I think that both the existing and proposal-2 methods both lie somewhere between super-professional and hackish. The new idea is only slightly more hackish than the other, so if it solves a practical problem, let's go for it. 

And thinking back to that [astronomy format ASDF](https://asdf-standard.readthedocs.io/en/stable/)you pointed me to a few years ago, this is basically the same. (I think they had YAML instead of JSON in the text header, but conceptually we are not far off from what a large consortium of experienced people was designing for huge data sets with thousands of potential users.)

ggggggggg, 2019-02-19:
I still think this is a good idea, but it does not take 10s of seconds, more like 2 seconds.

PR #151 (2019-05-22): write projectors and basis as binary after JSON header in off files
fixes #128 

the `fixes_for_ebit` branch of mass has corresponding changes to reading off files
this should speed up parsing off files, and make the headers much more human readable since there will no longer be a giant wall of random text taking up most of the header
closed 2019-05-22 by joefowler