usnistgov/dastard #130: Microwave MUX sources need to unwrap phase

joefowler, 2019-03-25:
The `RoachSource` and the still-in-developement `AbacoSource` need to unwrap phase by some means, to prevent ridiculous phase jumps. See what Dan Becker did in NDFB server, and also consider "asymmetric unwrapping", where large positive changes are considered more likely than mid-sized negative ones.

joefowler, 2019-04-12:
I don't understand this comment at all. Sorry. You're proposing what? That we attempt to recognize when the phase-unwrap algorithm has erred, by looking for ... what? For long runs that seem wrong? 

I think "nominal correct phase value" is throwing me. Do you mean, the typical baseline level, to which we expect the readout to return sooner or later?

ggggggggg, 2019-04-12:
Yeah, basically if you are far away from the "correct phase" for a long time, you add subtract a phi0. Pulses without unlocks only take you away from the "correct phase" for some number of samples, unlocks will make it so you have long runs of values far from the "correct phase".

joefowler, 2019-04-12:
Just discussed this with Doug. He is not convinced. Thinks we need to discuss Kelsey's current approach with her before taking any action along the lines of an automatic adjustment. At any rate, that would presumably be a separate issue.

ggggggggg, 2019-04-17:
ndfbserver has a detailed comment about how it unwraps: https://bitbucket.org/nist_microcal/nasa_daq/src/3eed1a3bb8c1e0270d3a2b8201fa19e9beeba000/ndfb_server/RoachUDPProcessor.cpp#lines-25

ggggggggg, 2019-05-09:
Here is the implementation of biased unwrap from ndfb_server on 687pegasus

```
        ////*** Used for biased unwrap for pulses
        //// Unwrap. last_val and tot are instance variables defined and
        //// initialized in header file, so they persist across packets.
        if ((val - last_val[ch]) >= (int16_t)(0.88*two_equiv)) {
            tot[ch] -= two_equiv;
        } else if ((val - last_val[ch]) < (int16_t)(-0.12*two_equiv)) {
            tot[ch] += two_equiv;
        }
```

ggggggggg, 2019-05-09:
Here is a hardcoded version of biased unwrap for Dastard 

```
	bias := 50
	posThresh := (int(twoPi) * bias) / 100
	negThresh := -(int(twoPi) * (100 - bias)) / 100
	//phi0_lim := (4 * (1 << (16 - bits_to_keep)) / 2) - 1
	for i, rawVal := range *data {
		v := int16(rawVal) >> bitsToShift
		delta := v - u.lastVal

		// short term unwrapping
		if delta > int16(posThresh) {
			u.offset -= twoPi
		} else if delta < int16(negThresh) {
			u.offset += twoPi
		}
```

But probably the better way to do it is to add `posThresh` and `negThresh` to `PhaseUnwrapper` as instance variables.

ggggggggg, 2020-01-16:
already done, though likely to be revisted by #177

closed 2020-01-16 by ggggggggg