usnistgov/dastard #133: Dastard should load Nsamp and Lsync from the file written by cringe

ggggggggg, 2019-04-09:
See this commit for Cringe details https://bitbucket.org/nist_microcal/nist_lab_internals/commits/82ebef0f214931257cb442f6ae6fa870cdde3705

joefowler, 2019-04-12:
When should Dastard read this file? On startup seems fine but insufficient. What if the values change while Dastard is running?

We could read when data acquisition on the Lancero device begins, but what if the values conflict with the configuration sent by the dastard-commander?

Or perhaps you are proposing that we never treat these values as user-configurable but always demand that they appear in `${HOME}/.cringe/cringeGlobals.json`?

ggggggggg, 2019-04-12:
I see the choice like this

A. We want to preserve the ability to determine these values the existing way:
  * Add an rpc call `readCringeGlobals` and a button in Dastard Commander that calls this. Or maybe a checkbox to call this before startup.
B. We never treat these values as user-configurable
  * Read from `${HOME}/.cringe/cringeGlobals.json` on dastard startup.

A feels safer to me, I can't explain why, but I worry that we would regret B at some point.

joefowler, 2019-04-12:
I share your misgivings about B. I'd modify B, however, to say that we read the config file as part of the Lancero start procedure (on the theory that cringe might have changed these values in the very recent past), not when Dastard starts.

On the other hand, A is way more work and adds needless complexity to Dastard and DC both. Perhaps if we have concerns about the safety of B, we should discuss and figure out if they are real or imaginary.

ggggggggg, 2019-04-12:
Here is one example.... autotune can set nsamp to 1 while the lancero source is running. It's done here: https://bitbucket.org/nist_microcal/nist_lab_internals/src/82ebef0f214931257cb442f6ae6fa870cdde3705/viper/cringe/tune/tunetab.py#lines-731

But NDFBServer or Dastard never know it was changed, and it gets changed right back.

I guess if you really wanted to force specific values you could just overwrite the `.json` file.

joefowler, 2019-04-12:
Right. I was forgetting that user configures Nsamp, but Dastard uses timing to estimate Lsync.

Proposal:
* User-configuration of Nsamp is moved to the "experts menu" of the GUI. (Do we still have that?)
* GUI offers "Read stored value" as a possible Nsamp value, indeed the default.
* Dastard's Lancero configuration object can treat Nsamp=0 as this default.
* We continue to estimate Lsync and give a warning if it doesn't match the stored value, but we always accept the stored value as true.

I don't understand your example. Are you saying that we want Server/Dastard to know that Nsamp was briefly set to 1, or not to know? I assume it's fine that they never know, and my proposal would work that way.

ggggggggg, 2019-04-12:
In my example Server/Dastard never know that NSAMP was changed. Dastard Commander doesn't have an experts menu, so a slightly modified proposal:
  * Nsamp in dastard commander is set to read only in the GUI, maybe we make it a label instead of a spin to make that obvious.
  * When you start a lancero source, dastard reads the cringe json file to learn nsamp and lsync.
  * Lancero estimates lsync and gives a warning if it doesn't match the stored value, but we always accept the stored value as true.

joefowler, 2019-04-12:
In your proposal, it sounds like the GUI doesn't have any control over Nsamp. You okay with that?

If it's a QLabel instead of a spin, then maybe we keep it blank until you first start Lancero; and maybe we gray it out after you stop Lancero (don't want to hide, just b/c user might want to know what the value was).

ggggggggg, 2019-04-12:
Yeah, if somebody really wants a specific nsamp in the server and a different one on the crate they can
1. overwrite the json file
or 
2. change nsamp in cringe after starting dastard

joefowler, 2019-04-29:
Fixed by #138.

closed 2019-04-29 by joefowler