usnistgov/dastard #134: Update travis config to use xenial (Ubuntu 16.04)

joefowler, 2019-04-11:
Our readme says Ubuntu 16 is required, but our Travis testing uses 14. Hmm.

ggggggggg, 2019-04-12:
I think it's really dastard commander that requires Ubuntu 16... its like python2.7 qt5 is hard to get on ubuntu 14.

joefowler, 2019-04-12:
Ah! Right. But in that event, we should at least configure Travis to test on Xenial instead of Trusty (16 instead of 14). I vaguely remember that there's a problem with loading the required packages that we never figured out on 16. I will explore. If I can't make it work, I'll put this on hold.

According to [Travis documentation](https://docs.travis-ci.com/user/reference/overview/), they support only Ubuntu 12, 14, 16, and 14 is still the default. That means that 18 isn't a choice yet.

PR #135 (2019-04-12): Travis: use Linux 16.04 instead of 14.04
This changes the Travis-CI configuration to build on Xenial, Linux 16.04 instead of 14.04. This happens to simplify the setup (because packages `libzmq3 libzmq3-dev libsodium` are all now part of the base line. It simplifies the setup _and_ speeds it up by around 60 seconds, actually.

I'm going to merge into branch:abaco at the moment, but that should be into master pretty soon, too.

Fixes #134
closed 2019-04-12 by joefowler