usnistgov/dastard #136: Gonum mat.Dense is being used as a concrete value rather than a pointer

kortschak, 2019-04-13:
In a number of places you have functions taking parameters that are `mat.Dense`. The `mat.Dense` type should be passed as a pointer value. You are currently not doing anything that will break because of this, but you are opening up yourself to problems if people forget.

I was going to send a PR to fix this, but you have an interface that locks this in place, so I figured it would be better to let you deal with it.

ggggggggg, 2019-04-17:
Thanks for the tip, I'll look into fixing it.

ggggggggg, 2019-05-09:
@kortschak #150 should fix this, would you mind taking a look? Now I pass all arguments as `*mat.Dense` and instantiate local variables and fields as `mat.Dense`.

PR #150 (2019-05-14): function arguments now *mat.Dense not mat.Dense
fixes #136
closed 2019-05-14 by ggggggggg