usnistgov/dastard #140: Should not panic when cringeGlobals.json doesn't exist

joefowler, 2019-04-29:
I think it's a mistake to panic on an absent `cringeGlobals.json` file. It's fine to say that the Lancero source is unusable in this case, but I think we are just begging for confused questions from users if we panic.

I'll think about whether disabling Lancero sources is easy to do.... See PR #138 and the issue #133 for more information.

On my mac:

```
fowlerj@817jfmac:dastard$ go run cmd/dastard/dastard.go 
2019/04/29 09:17:39 cringeGlobalsPath /Users/fowlerj/.cringe/cringeGlobals.json
2019/04/29 09:17:39 Dastard is using config file /Users/fowlerj/.dastard/config.yaml
2019/04/29 09:17:39 ConfigureSimPulseSource: 2 chan, rate=6000.000
2019/04/29 09:17:39 Result is okay=true and state={2 chan, rate=6000.000}
2019/04/29 09:17:39 ConfigureTriangleSource: 1 chan, rate=10.000
2019/04/29 09:17:39 Result is okay=true and state={1 chan, rate=10.000}
2019/04/29 09:17:39 ConfigureLanceroSource: mask 0xffff  active cards: []
2019/04/29 09:17:39 Result is okay=false and state={0 MHz clock, 0 cards}
panic: open /Users/fowlerj/.cringe/cringeGlobals.json: no such file or directory

goroutine 1 [running]:
github.com/usnistgov/dastard.RunRPCServer(0x157c, 0x4707a01)
	/Users/fowlerj/go/src/github.com/usnistgov/dastard/rpc_server.go:600 +0x997
main.main()
	/Users/fowlerj/go/src/github.com/usnistgov/dastard/cmd/dastard/dastard.go:96 +0x1bd
E: 19-04-29 09:17:40 [8859]dangling 'PUB' socket created at :179
exit status 2
```

ggggggggg, 2019-04-29:
When did this happen? it shouldn't look for this file in testing? But I think you're right since the configure function is called on loading the viper config files, that it would panic when sombody only cares about abaco or Roach. If it only paniced when you tried to start the lancero source it would be ok. The simplest fix I can think of it to pass an extra flag to `ConfigureLanceroSource`, a bool with a name like `RequireCringeGlobalsJSON` that would be `false` when loading from viper and `true` when starting the source.

joefowler, 2019-04-29:
This happened when I (tried to) start dastard. It wasn't when I tried to start a Lancero source. (Indeed, I would not be able to try, because there are no `/dev/lancero_user*` device files on my Mac.)

I forgot to mention this before: right now, users all have Lancero devices, and the only user lacking a cringe file would be me. But in the future, there will be µMUX users who have never run cringe on their lab machine.

joefowler, 2019-04-29:
Alternative to your idea: can we let `ConfigureLanceroSource` succeed without the cringe config file, but we somehow use settings that leave the source marked as do-not-use? This would be a simple way to make life easy for µMUX users.

ggggggggg, 2019-04-29:
Yes, that's also possible. At `rpc_server.go` line 313 is checks the call to `Start` for an error, so if we just make `Start` error we should be fine. But `ConfigureLanceroSource` must be called before `Start` over the RPC seperatley, and thats when we know if we read the file. So maybe add a field to `AnySource` like `errorDuringConfigure` and check it in `Start`?

joefowler, 2019-04-29:
That makes sense. I will see what I can do....

commit 9bbb9b3e (2019-04-29): Failure to ConfigureLanceroSource does not cause a panic

If `ConfigureLanceroSource` fails (e.g., because there's no cringe configuration file), then remember this error but don't panic. For a non-Lancero computer, this failure is normal behavior.

Instead of panicking, remember the error and refuse to start the source in the `Sample()` method. (Given that a non-Lancero machine won't have `/dev/lancero_user*` files, this will not actually come up in practice.)

Fixes #140.
PR #141 (2019-04-29): Failure to ConfigureLanceroSource does not cause a panic
If `ConfigureLanceroSource` fails (e.g., because there's no cringe configuration file), then remember this error but don't panic. For a non-Lancero computer, this failure is normal behavior.

Instead of panicking, remember the error and refuse to start the source in the `Sample()` method. (Given that a non-Lancero machine won't have `/dev/lancero_user*` files, this will not actually come up in practice.)

Fixes #140.
closed 2019-04-29 by joefowler