usnistgov/dastard #143: Triggers are buggy with Abaco source

joefowler, 2019-04-29:
If I run the Abaco data source, I get many many of the following terminal messages per second. I don't know why they appear, but they are a problem. The message is nearly infinitely long, so I am showing only the end of one:

```text
...
  (dastard.FrameIndex) 72490100,
  (dastard.FrameIndex) 72490500,
  (dastard.FrameIndex) 72490900,
  (dastard.FrameIndex) 72491300,
  (dastard.FrameIndex) 72491700,
  (dastard.FrameIndex) 72492100,
  (dastard.FrameIndex) 72492500,
  (dastard.FrameIndex) 72492900
 },
 keyFrame: (dastard.FrameIndex) 72258160,
 keyTime: (time.Time) 1727-01-18 16:15:38.456383084 -0659 LMT,
 sampleRate: (float64) 0,
 lastFrameThatWillNeverTrigger: (dastard.FrameIndex) 72493268
}

(dastard.TriggerCounter) {
 channelIndex: (int) 0,
 hi: (dastard.FrameIndex) 72258160,
 lo: (dastard.FrameIndex) 71987313,
 hiTime: (time.Time) 1727-01-18 16:20:29 -0659 LMT,
 countsSeen: (int) 0,
 stepDuration: (time.Duration) 1s,
 sampleRate: (float64) 0,
 keyTime: (time.Time) 1727-01-18 16:15:38.456383084 -0659 LMT,
 keyFrame: (dastard.FrameIndex) 72258160,
 initialized: (bool) true,
 messages: ([]dastard.triggerCounterMessage) (len=1 cap=1) {
  (dastard.triggerCounterMessage) {
   hiTime: (time.Time) 1727-01-18 16:20:28 -0659 LMT,
   duration: (time.Duration) 1s,
   countsSeen: (int) 0
  }
 }
}

```

joefowler, 2019-04-29:
@ggggggggg: I feel like we've discussed this problem, but I can't find any old issues that appear to be relevant. Hmm.

ggggggggg, 2019-04-30:
This points to something like the `FrameIndex` being non-monotonic, or the differences in `FrameIndex` are not equal to the number of samples.

joefowler, 2019-04-30:
Yep, it must something like that. It also concerns me that `hiTime` and `keyTime` are set to January of 1727. That might be wrong.

I wonder if dastard is assuming an incorrect frame rate that turns seconds into 3 centuries? (That's an error of 9-10 orders of magnitude.) I'll investigate.

ggggggggg, 2019-04-30:
I didn't notice that, that's a definite red flag.

joefowler, 2019-05-03:
Okay, I was taking `AbacoSource.sampleRate` to be 0 samples per second. Duh. For now, we'll just assume a value. Will need to discuss with Mazsi and John how to get a smarter value...

ggggggggg, 2019-05-03:
I agree, just hardcode in the correct rate for the first few tests and come up with a better soluiton eventually.

commit ae4470f2 (2019-05-03): Set a fake, but non-zero sampleRate for AbacoSource

Eventually will want to read this from the raw data, but this is a start. Also move where enumerateAbacoDevices() lives in the file. Fixes #143
closed 2019-05-03 by joefowler