usnistgov/dastard #162: map info passed to file writers is wrong when error channels exists

ggggggggg, 2019-07-15:
The relevant code is here:
https://github.com/usnistgov/dastard/blob/2e832a37a831645b4b112b9ea6ed3cf2e6c67a57/data_source.go#L506-L513

This code needs to account for the fact that there can be more than on `channelIndex` associated with a signal channel. Something like:

```go
pixel = config.m.Pixels[i/channelIndiciesPerSignalChannel]
```

and before the loop

```go
channelIndiciesPerSignalChannel = howDoIGetThisInfo()
if config.map != nil {
    if len(config.m.Pixels) != len(ds.processors) {
        return fmt.Errorf("map file wrong length")
        }
    }
```

I think returning an error here is the right thing, before I tried to gracefully handle the missing info and just did it wrong. It's not really useful to show a map that's not accurate anyway.

ggggggggg, 2019-07-15:
Work in progress on `map_info_to_ljh3 ` branch. 

Assume the map file is wrong, what should the user do? They either need to load a correct length map file, but that might not be immediately available. So they should `unload` the map file, so we need an unload operation. Or we could have an option in file writing `StrictMap` that controls behavior on incorrect map, either returning an error or populating the map with dummy info.

Also in the tests there is currently a map file loaded with 240 channels, so some tests are failing the strict comparison when they only have 4 channels. With `StrictMap` option I could just let the tests ignore the 240 channel map. Or I could make a 4 channel map for tests to load.

joefowler, 2019-09-05:
Fixed by #165.

closed 2019-09-05 by joefowler