usnistgov/dastard #164: Abaco can't keep up with high-rate data

joefowler, 2019-08-14:
Abaco source can keep up with 133 MB/sec data just fine, but at 2x this, one Goroutine begins to use > 100% CPU  and Dastard eventually crashes.

``htop`` suggests that all the CPU is in one goroutine, probably (?) the one that does de-MUXing. Two ideas to share that work:

1. Launch 2 goroutines to do the deMUXing, perhaps odd vs even channels. If that splits the heaviest load into 2 goroutines, then we know where the bottleneck is. Then start launching even more than 2. Or...
1. Don't do de-MUXing with a copy. Just pass a pointer and a stride to the downstream processing goroutines, and wait for all N of them to announce their completion. Once they are all done, we can release the relevant section of the `RingBuffer`. Mazsi advocates this one.

ggggggggg, 2019-08-14:
The first order of buisness should be to confirm that it is actually the demux step, perhaps by using print statements to measure what fraction of time is spent demuxing. 

Additional options:
  3. Look into Young's vectorized (SIMD, AVX) demux code and getting it to run in go. We would need to use cgo or go assembly to implement it. We already use cgo (`import "C"`) in two lancero related files? He told me it was ~5x faster IIRC.
  4. Allow demux to happen on a different thread from data reading. Roughly we would copy the data, then launch a goroutine to demux the data and put it on a channel. We may need to be careful about the order of demux completion.

jojo978, 2019-08-15:
Actually, I only tried a SIMD version of mixing. And I think it helped. But
I found it's not so easy to implement the demuxing (of ndfb server) using
SIMD instruction.

I guess it would be a dedicated hardware to demux without intervention of
CPU. Though I don't know exactly how it could be.

ggggggggg, 2019-08-15:
Here is a stack overflow thread about demultiplexing quickly, it may have useful info: https://stackoverflow.com/questions/41138596/need-for-fast-data-demuxing-in-c-sharp-by-using-multi-threading-avx-gpu-or-wha

ggggggggg, 2019-08-15:
Here is a simple demux speed test file, I consistently get rates of about 250 MB/s on my macbook air. 

```go
package main

import (
	"fmt"
	"time"
)

func main() {
	tstart := time.Now()
	var nb int
	for k := 0; k < 1000; k++ {
		nchan := 8
		nsamples := 100000
		int32Buffer := make([]int32, nchan*nsamples)
		datacopies := make([][]int16, nchan)
		for i := 0; i < nchan; i++ {
			datacopies[i] = make([]int16, nsamples)
			dc := datacopies[i]
			idx := i
			for j := 0; j < nsamples; j++ {
				dc[j] = int16(int32Buffer[idx] >> 12)
				idx += nchan
			}
		}
		nb += nsamples * 4
	}
	tend := time.Now()
	elapsed := tend.Sub(tstart)
	fmt.Printf("%v bytes in %v seconds, %v MB/s\n", nb, elapsed, float64(nb)/elapsed.Seconds()/1e6)

}
``

joefowler, 2019-08-15:
Thanks, Young. I think we forgot that the SIMD was for mixing rather than deMUXing.

Galen, I'm not sure what to conclude from your test. 250 MB/s is not what we are hoping to achieve. That's the current rate, but we are trying to make sure we can handle more like 500 MB/s. Once we can, we will declare success and move on.

One thing I learned yesterday is that we make 2 copies of the data:
1. From the shared-memory ring buffer to a Golang buffer. This has the advantage that if we read a portion of the ring buffer that's discontinuous in the ring, then only the copy step needs to know that.
2. From the single Golang buffer to the per-channel buffers. This is the deMUX step.

So the slow single-threaded bit could be in any or all of the following:
1. The first copy.
2. The second copy / deMUX step.
3. Downstream processing (everything after deMUX but before each TES data packet goes to a separate goroutine).

Galen's Stack Overflow link suggests that we are not doing the deMUX in the optimum way, at least not from the point of view of cache hits on _reading_ the data. A test I did yesterday suggests that either item 2 or 3 is the problem, rather than the first copy. I will attempt to refine the test to discriminate all three cases.

ggggggggg, 2019-08-15:
Joe my conclusion from my demux test script is that the demux is probably the problem given that I can only demux at 250 MB/s, which is vary similar to the data rate at which dastard fails to keep up.

ggggggggg, 2019-08-20:
We did some tests with actual hardware where we found that the PhaseUnwrap step was an important cause of CPU usage. I think we got up to about 380MB/s when we removed or multithreaded the phase unwrap. The demux may be the limiting factor at that rate.

joefowler, 2019-08-20:
@ggggggggg suggested the idea of doing deMUX with some BLAS matrix-transpose routine. I agree that matrix-transpose is what we're doing. Let's look into it.

See [OpenBLAS issue 50](https://github.com/xianyi/OpenBLAS/issues/50), which oddly enough was opened by @ViralBShah in 2011 when Julia was still a baby. Looks like OpenBLAS has had a copy and copy-transpose API for several years. See `?imatcopy` for in-place transposition and `?omatcopy` functions for out-of-place transpose. [OpenBLAS extensions](https://github.com/xianyi/OpenBLAS/wiki/OpenBLAS-Extensions) describes them.

ggggggggg, 2019-08-21:
BLAS doesn't include a transpose operation, I think because most operations allow you to tell blas to treat one or more matricies as if they're transposes, so you never need to instantiate the transposed matrix. Thus, as far as I can tell, there is no non-lazy transpose routine in gonum.org/mat. According to this stackoverflow: https://stackoverflow.com/questions/7784651/how-to-transpose-a-matrix-in-an-optimal-way-using-blas cublas (cu means graphics card I think) has a transpose operation called `geam`, and there is this cublas package for golang: https://godoc.org/gorgonia.org/cu/blas so maybe we could cublas for transpose? The golang cublas package has no documentation, and has functions called `Cgeam`, `Dgeam`, etc, so it would take more research to figure out what to actually call.

ggggggggg, 2019-08-21:
I asked on gonum-dev how to do a non-lazy transpose, and was pointed to 
```
var mat Dense
mat.CloneFrom(b.T())
```
which leads to this code:
https://github.com/gonum/gonum/blob/d61003946d0d62fea7df4d9f9dee3c729ff97c9e/mat/dense.go#L397

joefowler, 2019-08-21:
Well, `Cgeam Dgeam` have the usual [BLAS prefixes](https://www.rpi.edu/dept/cis/nic/doc/blas/blas). 

```text
Meaning of prefixes:
 
   S  - REAL or REAL*4
   D  - DOUBLE PRECISION or REAL*8
   C  - COMPLEX or COMPLEX*8
   Z  - COMPLEX*16
   EC - COMPLEX*8 with extended precision accumulation
   ES - REAL*4 with extended precision accumulation
 ```

So as long as we are getting 32-bit data from Abaco, we can use any existing S-prefix code.  Notice that the CUBLAS package does have a series of S-prefix functions. It doesn't have `SCOPY`, but if you look inside the Go package, you'll see how they wrap and call the underlying C library. Maybe we could just do that for the one function of interest.

One problem with any `gonum/mat` solution is that the package handles only 64-bit floats. The float part is no problem, as long as we're only copying. But the 64-bit would mean that after any `mat` transpose, we'd still have channels (say) 1 and 2 interleaved. Any savings from the efficient transpose could end up squandered on a second pass to deMUX pairs. (Plus we'd either have to handle or forbid a case where there's an odd number of channels. Not cool.)

In short, I'm inclined to look into a direct library call, either `scopy` in CUBLAS or the OpenBLAS extension I mentioned before, `somatcopy`, I think. Surely one of these is available on Linux?

ggggggggg, 2019-08-21:
I realized we can also do this in deed, which is in c, so it might be easier. And if we do packetization, I'm pretty sure they can just demux during packetization without much cost, but Mazhi would know for sure.

joefowler, 2019-08-22:
Reading about high-performance computing, I discovered this highly relevant wisdom:

“A supercomputer is a device for turning compute-bound problems into I/O-bound problems.”
-- Ken Batcher, Emeritus Professor of Computer Science at Kent State University

joefowler, 2019-08-23:
I'm going to close this issue, and we can handle the deMUX step as a separate issue, if needed. It's possible that we won't need to, because Mazsi is thinking about deMUXing data in chunks of approximate size 8 kB (i.e., deMUXing a packet at a time). That might be enough to solve our speed problems. If not, it's a separate issue.

closed 2019-08-23 by joefowler