usnistgov/dastard #191: configurable channel names (number) offset

ggggggggg, 2020-05-07:
Looking forward to running 6 abaco cards on three computers, we may want to have unique channel names among all channels across those three computers. One way to achieve this would be to allow dastard to have a channel number offset, so channel numbers start at the offset an increase from there. By having a different offset on each computer, we could then have unique channel numbers across all channels.

This code is relevant:
https://github.com/usnistgov/dastard/blob/cca7cf1c4eb608f21888959736f57f351877d92c/data_source.go#L692-L705

joefowler, 2020-08-20:
I definitely see the appeal of this. And of course, we must have channel numbering be globally unique!

But I would much prefer to see this globally unique system imposed from the beginning of the DAQ chain, so that each Abaco card has its own global offset, _known to the card's firmware_. In this way, channel 10234 or whatever will be a single resonance, numbered the same during autotuning, Dastard operation, and analysis. It would be a serious pain to have to translate "card 2 channel 234" at tune-up time into "channel 490" at Dastard time (because 490=234+(2-1)*256).

In this world, Dastard would never change a channel number, but it would merely report and repeat the pre-assigned numbers. It would be on the Abaco managers to make sure that numbers are unique. At the moment, Dastard panics if it sees channel groups 0-7 and 0-15. We could add a panic if it sees 0-7 from more than one ring buffer, I guess.

I propose one of the following:
1. Mark this issue wontfix and expect unique-ification to be done in firmware.
2. Implement an offset and plan for it to be set to zero in almost all foreseeable situations.

I favor proposal 1, but there is one advantage to proposal 2: we could use the offset in a TDM experiment. Do we expect any TDM experiments to use 2+ PCs? I should probably ask Randy that question.

Last point: we discussed yesterday that we should check through the Dastard code base overall and make sure that we have a correct handling of non-consecutive channel numbers. This step might solidify the distinction between channel _number_ (a label) and channel _index_. The former would be public. The latter would be hidden, internal to Dastard, and run from 0 to N-1. If we have to make big changes for this, then a configurable channel offset might be a trivial addition.

joefowler, 2020-09-10:
I got M on the firmware team to agree today that we can initially plan on proposal 1, above: firmware will be configured to have globally unique channel numbers in an Abaco-based experiment. In the event that we ever have to handle multiple TDM systems, then we can add this as a feature, possibly only to the `LanceroSource`.

For now, I will consider this issue complete, except for the thorough check about channel numbers vs indices.

PR #213 (2020-09-11): Channel numbering and indexing are separate concepts.
I've gone through objects `AnySource`, `DataSource`, `TriggerBroker` and `DataStreamProcessor` to make sure that nothing therein assumes that channels are numbered consecutively. Also removed the calculation of "channels per pixel" (which would have been broken by the non-consecutive numbering) and delegate that to the source, so that we generally get an answer of 1 except that `LanceroSource` has 2. Fixes #191.

Will still need to add features so Lancero sources can be non-consecutive (see #212), and fix Microscope to use real channel numb [... cut]
closed 2020-09-11 by joefowler