usnistgov/dastard #192: repeated zmq errors

ggggggggg, 2020-05-11:
Looking at the output of dastard over a few days I see many (100?) messages like the following 

```zmq4: could not open a ZMTP connection with "tcp://*:5502": zmq4: could not initialize ZMTP connection:
    github.com/go-zeromq/zmq4.Open
        /home/pcuser/go/pkg/mod/github.com/go-zeromq/zmq4@v0.9.0/conn.go:92
  - zmq4: could not exchange greetings:
    github.com/go-zeromq/zmq4.(*Conn).init
        /home/pcuser/go/pkg/mod/github.com/go-zeromq/zmq4@v0.9.0/conn.go:104
  - zmq4: could not recv greeting:
    github.com/go-zeromq/zmq4.(*Conn).greet
        /home/pcuser/go/pkg/mod/github.com/go-zeromq/zmq4@v0.9.0/conn.go:146
  - could not read ZMTP greeting:
    github.com/go-zeromq/zmq4.(*greeting).read
        /home/pcuser/go/pkg/mod/github.com/go-zeromq/zmq4@v0.9.0/protocol.go:98
  - read tcp 132.163.82.105:5502->132.163.5.35:41178: read: connection reset by peer
```

@sbinet What does this mean? Does this correspond to a line in dastard (if so, how can I find out which one) or is this an internal message from `go-zeromq`?

sbinet, 2020-05-11:
This is the display of an error originating from zmq (but I suspect it's being displayed by dastard).
This error happens when opening a new zmq connection and the peer dropped off during the zmq greeting handshake.

ggggggggg, 2020-05-11:
Thanks. I suspect this is something the previous zmq library just didn't show us, and it is due to a security scanner trying to connect in some way. Is there a way to suppress the output?

sbinet, 2020-05-11:
there are 2 places where `dastard` is creating `zmq4.Socket`s:
- https://github.com/usnistgov/dastard/blob/9b4ceeecc4d331502713b7007e24833e57da5b3c/client_updater.go#L49..L52
- https://github.com/usnistgov/dastard/blob/9b4ceeecc4d331502713b7007e24833e57da5b3c/publish_data.go#L378..L385

if it's not coming from there, then it's indeed coming from zmq4, here:
- https://github.com/go-zeromq/zmq4/blob/de8318e9c31dc075627022e399766a39e0b5e2aa/socket.go#L201

then, a simple "fix" would be to create a `/dev/null` logger:
```go
pubSocket := zmq4.NewPub(
    context.Background(), 
    zmq4.WithLogger(log.New(ioutil.Discard, "dastard: ", 0)),
)

ggggggggg, 2020-05-11:
Thanks. I think I'll leave it for now, since it doesn't actually break anything. Maybe I'll have it log to a `dastard_zmq.log` file in the future.

joefowler, 2020-05-26:
Fixed by changing ZMQ packages (#193).

closed 2020-05-26 by joefowler