usnistgov/dastard #200: Bahama-generated data causing trigger/timing problems

joefowler, 2020-07-28:
Problems with the Abaco data source when Bahama is generating the data. When we run it and have triggers, the dastard terminal goes crazy like the following. We are also generating something like 10 TRIGGERRATE messages per second, but Galen thinks that is a consequence of the same bug that makes the error messages happen.

```text
2020/07/28 10:02:01 triggering assumptions broken!
frame 2135000 still higher than tc.hi=2125000 after reset (Δf=10000)
(dastard.triggerList) {
 channelIndex: (int) 5,
 frames: ([]dastard.FrameIndex) (len=1 cap=1) {
  (dastard.FrameIndex) 2135000
 },
 keyFrame: (dastard.FrameIndex) 2125000,
 keyTime: (time.Time) 1728-04-18 09:14:48.580232192 -0659 LMT,
 sampleRate: (float64) 0,
 lastFrameThatWillNeverTrigger: (dastard.FrameIndex) 2150000
}

(dastard.TriggerCounter) {
 channelIndex: (int) 5,
 hi: (dastard.FrameIndex) 2125000,
 lo: (dastard.FrameIndex) 2105001,
 hiTime: (time.Time) 1728-04-18 09:16:12 -0659 LMT,
 countsSeen: (int) 0,
 stepDuration: (time.Duration) 1s,
 sampleRate: (float64) 0,
 keyTime: (time.Time) 1728-04-18 09:14:48.580232192 -0659 LMT,
 keyFrame: (dastard.FrameIndex) 2125000,
 initialized: (bool) true,
 messages: ([]dastard.triggerCounterMessage) (len=1 cap=1) {
  (dastard.triggerCounterMessage) {
   hiTime: (time.Time) 1728-04-18 09:16:11 -0659 LMT,
   duration: (time.Duration) 1s,
   countsSeen: (int) 0
  }
 }
}
2020/07/28 10:02:01 triggering assumptions broken!
frame 2135000 still higher than tc.hi=2125000 after reset (Δf=10000)
(dastard.triggerList) {
 channelIndex: (int) 6,
 frames: ([]dastard.FrameIndex) (len=1 cap=1) {
  (dastard.FrameIndex) 2135000
 },
 keyFrame: (dastard.FrameIndex) 2125000,
 keyTime: (time.Time) 1728-04-18 09:14:48.580232192 -0659 LMT,
 sampleRate: (float64) 0,
 lastFrameThatWillNeverTrigger: (dastard.FrameIndex) 2150000
}

(dastard.TriggerCounter) {
 channelIndex: (int) 6,
 hi: (dastard.FrameIndex) 2125000,
 lo: (dastard.FrameIndex) 2105001,
 hiTime: (time.Time) 1728-04-18 09:16:12 -0659 LMT,
 countsSeen: (int) 0,
 stepDuration: (time.Duration) 1s,
 sampleRate: (float64) 0,
 keyTime: (time.Time) 1728-04-18 09:14:48.580232192 -0659 LMT,
 keyFrame: (dastard.FrameIndex) 2125000,
 initialized: (bool) true,
 messages: ([]dastard.triggerCounterMessage) (len=1 cap=1) {
  (dastard.triggerCounterMessage) {
   hiTime: (time.Time) 1728-04-18 09:16:11 -0659 LMT,
   duration: (time.Duration) 1s,
   countsSeen: (int) 0
  }
 }
}
```

ggggggggg, 2020-07-28:
Where is the bahama code and how does one test it?

joefowler, 2020-07-28:
Branch:bahama. You need a terminal to run it. My session looks like this:
```text
fowlerj@687figaro:dastard$ go run cmd/abaco/bahama.go -noise 50 -saw -nchan 8 -nring 1
nchan:  8
nring:  1
rate:   10000
Data will contain: noise sawtooth.
Generating data in shm:xdma0_c2h_0_buffer
```
It blocks at the above and does not print periodic "I am okay" messages. There is a usage message if you need it. Use the `go run cmd/abaco/bahama.go -h` to get it.

You also need to run Dastard and Dcom as usual. (That is, Bahama doesn't replace Dastard.)

joefowler, 2020-07-28:
The fact that my timestamps spew as the year 1728 is surely a clue...

ggggggggg, 2020-07-28:
Highly suspicious. Thats even before Hamliton takes place.

joefowler, 2020-07-28:
Yes, 1728 was the year that Governor George Phenney reformed the governance of the Bahamas, leading to the first democratic elections in the following year. Do you think that might be relevant?

joefowler, 2020-07-28:
Okay, the Bahama-generated packets have null timestamps. Or at least some of them do when we are in the sample-data phase (i.e., starting the Abaco source):
```
(packets.PacketTimestamp) {
 T: (uint64) 0,
 Rate: (float64) 0
}
```

I'm not sure what the effect is, but it's clearly causing problems.

commit 1d676c93 (2020-07-29): Bahama simulator now fills packet timestamps appropriately

Fixes #200.
closed 2020-07-29 by joefowler