usnistgov/dastard #207: there should be a "projectors" message so a gui can know if the projectors state changed

ggggggggg, 2020-08-28:


joefowler, 2020-09-08:
This would be easy enough, but does it contain any other information? Maybe it should say when the state changed? Is there anything else interesting to propagate? File name for the projectors, or anything like that?

ggggggggg, 2020-09-08:
In principle we could just include the projectors. I'm not sure what the best solution is... probably the right answer is just don't worry about how much data we're sending until there is an actual problem and just send out all the info we have including the actual projectors. But who should be responsible for knowing when the projectors changed if eg I stop a source? Doesn't that change the projectors as well since when we start a new source we will have no projectors loaded? Should dastard send a message in that case, or should guis know what is up?