usnistgov/dastard #217: handle unknown TLVs gracefully

jozsef-imrek-at-nist, 2020-10-13:
for packets with an unknown TLV, dastard seems to stop with error

> Oh no! error in ReadAllPackets: Unknown TLV type ..

unknown TLVs should be ignored gracefully.

ggggggggg, 2020-10-13:
I'm not up on my abaco lingo, what is TLV?

joefowler, 2020-10-13:
Totally Lame Value. It's how packets can carry arbitrary information in the header.

Type-Length-Value might actually be the name. @jozsef-imrek-at-nist and I discussed this last week, and I asked him to file as an issue. We agreed it was reasonable for Dastard to ignore unknown TLVs. If we invent some new, mandatory TLV type, then we can update the packet header version number, which would make older non-conforming versions of Dastard fail.

PR #219 (2020-10-20): Handle TAG-type TLVs. Ignore TLVs types that aren't handled.
Fixes #217. Fixes #216.
closed 2020-10-20 by ggggggggg