usnistgov/dastard #225: incorrect unit label on PSD and FFT

hubmayr, 2021-01-13:
Units are listed as mV^2/Hz or mV/sqrt{Hz}, but they are in fact micro Volts.

ggggggggg, 2021-01-14:
This could be an error on the dastard size or the microscope side, microscope is more likely.

hubmayr, 2021-01-15:
not sure if this is definitive, but writing my own script using easy client
does extract microvolts.  Since that uses dastard, doesn't that mean
microscope is the problem?

ggggggggg, 2021-01-15:
Probably, there place it could be
1. dastard sends out info about what the physical units are
2. microscope could make a mistake with that info

My guess is #2.

joefowler, 2021-01-29:
Hey @hubmayr. I am finally looking into this, and I cannot find the problem. Unfortunately, I'm not looking at any data where I know the physical power spectrum level. But I am looking at data where I am interpreting 65536 arbs = 1 Volt, or 1 arb = 15.3 µV. When I see a spectrum peaking at 13172 arbs/sqrt(Hz), it comes to 201 mV/sqrt(Hz). That all looks right to me.

So although I guessed (like Galen) that microscope was probably at fault, now I'm not so sure. Is that assumption 1 V = 2^16 arbs wrong for your case?

joefowler, 2024-11-14:
See https://github.com/usnistgov/microscope/issues/53 for possible cause within Microscope.

NanoExplorer, 2024-11-18:
I believe that this issue will be solved by usnistgov/microscope#53. I don't know what the experimental setup was when this was observed in 2021, but the Microscope issue caused it to report PSD numbers that were too large by a factor of about (3*sample rate). The Velma system can sample at up to about 300 kHz, which would cause 6 orders of magnitude offset in the PSD (3 in the FFT), which would look like the difference between mV^2 and uV^2.

joefowler, 2024-11-19:
Yes, @NanoExplorer seems to have it right. This has not been a Dastard problem.

closed 2024-11-19 by joefowler