usnistgov/dastard #239: failed to close experimentStatefile with Abaco

ggggggggg, 2021-04-08:
With the abaco source, if you use dcom to first start writing a file, then hit "stop writing", you get an RPC error originating from the line below and it appears to not actually stop writing.

https://github.com/usnistgov/dastard/blob/cc2d26404fd473b34123297ad9be3b68771b30aa/writing_state.go#L86-L88

ggggggggg, 2021-04-08:
@danbek

ggggggggg, 2021-04-08:
[image]

ggggggggg, 2021-04-08:
[image]

joefowler, 2021-04-08:
Yeah, that's pretty strange. If we already verified that `ws.externalTriggerFile != nil` then maybe we should add to that line 99 error message. We should print out `ws.externalTriggerFile`. Maybe it's an invalid file handle? Maybe we opened that file and failed to panic or recover from an error back at file-opening time?

Is this project actually using external triggers? Or is this an error that arises when we aren't trying to use them?

Who's running this, and where?

ggggggggg, 2021-04-08:
This is dan becker with abaco on 687medusa. It definitely does not have external triggers.

ggggggggg, 2021-04-08:
I added some debug printing, and then I couldn't reproduce the problem. I then removed the debug printing, recompiled, and cannot reproduce the problem. Maybe they were running an older version?

ggggggggg, 2021-04-12:
@danbek can still cause this

PR #240 (2021-04-13): Fix clear bug in `WritingState.Stop()`
Fixes #239. Also has test that failed before the fix.

I'm pretty sure this was just a copy-paste error when trying to duplicate code.
commit 5b8d3e85 (2021-04-13): Fix clear bug in `WritingState.Stop()`

Fixes #239. Also has test that failed before the fix.
closed 2021-04-13 by joefowler