usnistgov/dastard #242: subpar abaco UDP performance

ggggggggg, 2021-04-13:
Apparently the dastard side of the abaco+UDP story is to slow. Here is some research about high throughput udp in golang:

Reddit thread with lots of refs: https://www.reddit.com/r/golang/comments/kaacsz/udp_performance_is_suboptimal_in_go_true/
The reccomended function for low overhead UDP reading: https://golang.org/pkg/net/#UDPConn.ReadMsgUDP
example udp server test: https://gist.github.com/jtblin/18df559cf14438223f93

joefowler, 2021-04-13:
Thanks for the background reading! I also remember noticing some key step I'm doing in a less efficient way, but now I cannot remember it.

joefowler, 2021-05-20:
Okay, I now remember it. I am looping over the UPD reads where each (successful) read is decoded into a Dastard packet and put on a Go channel. But the channel reader only drains the channel every ~100 ms, which is no good, because we are receiving thousands of packets every 100 ms. Solution is clear now.

The references Galen found will be useful if we ever find that the UDP reading itself is a bottleneck. So far it appears not to be. The bottleneck was my dumb design.

commit 0d59bcf1 (2021-05-20): Redesign `AbacoUDPReceiver` for performance

Change design of `AbacoUDPReceiver` so packets are read and queued up in a new goroutine, which uses `select` to listen for a "send more data" request that sends all queued packets. Fixes #242.
closed 2021-05-25 by joefowler