usnistgov/dastard #247: Group triggering cannot be enabled

joefowler, 2021-05-27:
This is a feature we never completed. Although there's a `TriggerBroker` in [group_trigger.go](https://github.com/usnistgov/dastard/blob/master/group_trigger.go), we don't have yet any RPC commands to activate its features. HEATES will need it.
* Add RPC commands to manipulate the trigger broker.
* Store configuration of group trigger.
* Make it work.
* Fix #235 (I assume this is related?)
* In Dastard-commander, make GUI elements that can do elementary group triggering.

joefowler, 2021-12-18:
Also to do:

- [ ] Add some kind of crosstalk-like feature to BAHAMA so we can _test_ group triggering.

ggggggggg, 2022-03-16:
fixed by #264

joefowler, 2022-03-16:
This is absolutely _not_ fixed. All I did was to make FB->Error coupling work (in #264), which is a very specific special case.

PR #268 (2022-03-18): Group trigger is ready!
Fixes #247 finally. Features:
- Group triggers can connect any channel to any other.
- Request to trigger from X to itself is ignored.
- RPC commands to add and remove specific connections, or to remove all connections.
- The new trigger architecture (#264) is used.
- BAHAMA data simulator has a "crosstalk" mode where odd-numbered channels are 2% the usual level (useful for testing group trigger).
closed 2022-03-18 by joefowler