usnistgov/dastard #251: seffault during TestAbacoSource observed on 684tes

ggggggggg, 2021-10-01:
```
=== RUN   TestAbacoSource
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x57e3bd]

goroutine 65 [running]:
net.(*UDPConn).readFromUDP(0x47a592, {0xc000718000, 0x3, 0x0}, 0x3)
        /usr/local/go/src/net/udpsock.go:113 +0x1d
net.(*UDPConn).ReadFrom(0xc000202ae0, {0xc000718000, 0x2000, 0x2000})
        /usr/local/go/src/net/udpsock.go:125 +0x50
github.com/usnistgov/dastard.(*AbacoUDPReceiver).start.func3()
        /home/pcuser/go/src/github.com/usnistgov/dastard/abaco.go:483 +0x112
created by github.com/usnistgov/dastard.(*AbacoUDPReceiver).start
        /home/pcuser/go/src/github.com/usnistgov/dastard/abaco.go:479 +0x23f
```

joefowler, 2021-10-04:
Yeah, I also can't run tests without errors. I wonder what changed... My mac crashes somewhere in `goczmq` when I run tests or build the `dastard` binary and try to run it. That is definitely new.

joefowler, 2021-10-12:
See issues  [302](https://github.com/zeromq/goczmq/issues/302) and [279](https://github.com/zeromq/goczmq/issues/279) for `goczmq`.

I can't say that this problem you're describing here is a problem in `goczmq`, but the above 2 issues make it seem like that package is incompatible with contemporary Golang modules, and might no longer be actively maintained. (The earlier issue is 17 months old and has no obvious response.)

Can you say what version of library `czmq` is on your computer 684tes? I tried to log into it to check but don't know its password.

joefowler, 2021-10-12:
Actually, I retract my comments here. I think my problem is totally different from yours, which I cannot reproduce. It passes tests on 687ravdaq02. Is it possibly just an attempt to bind a socket that's already bound, and we are not checking an error where we ought to?

joefowler, 2022-03-16:
Now that we're clear what this issue is _not_, I can add that I definitely get that same error, maybe 10-20% of the time when I run tests on my machine. I don't know how to assess such an unusual error.

joefowler, 2022-05-14:
You mentioned this again today. You tend to skip certain tests to avoid it. 

An idea that might make it go away 99.9% of the time: I could try to have have the TEST version of the RPC server open up a port randomly between 50000 and 54999. If we use 5 consecutive ports, then the chance of a second test using any port from the first test's port range is 0.1%.

joefowler, 2022-05-17:
See 9514c9f2. I randomized the port number to be 30000, 30010, up to 39990, so if you start running a second test at the same time, chances are only 1/1000 that they collide. (The steps of 10 ensure that the 4 or so ports used by Dastard never halfway overlap, because that would be _really_ confusing.)

I have no idea if this really diagnoses the problem you raised, but I am going to hope so and close this issue.

closed 2022-05-17 by joefowler