usnistgov/dastard #254: WriteControl could/should return filename when starting

ggggggggg, 2021-12-14:
For `scan_server` (a python packge to expose the "TES API" to a beamline) we want to know where file are being written. Currently this is done asychronously, by first calling `WriteControl`. Then we listen for a message from dastard about writing having started. Jamie would prefer to remove the async part altogether, which could be done by stuffing some of the same info into the response to the command. Jamie has run into issues with the async response not showing up reliably, though I'm not sure 

Pros: 
Make an important thing easy and avoid some potentially error prone async programming.

Cons:
Doing the same thing more than 1 way, violating the basic principles of dastard design.

Thoughts?

joefowler, 2021-12-14:
Yeah, I don't think this idea is truly possible. I'm going on memory here, so I could be wrong, but I think that this concept is fundamentally incompatible with Dastard's design. The RPC reply is limited to little more than "yes/no I was/wasn't able to make sense of your request". The action itself is queued, so there's no way to get an immediate (synchronous) answer to stuff into the reply. The queueing of actions was the only way we managed to eliminate all the race conditions that we would otherwise have introduced. 

Alternative: Would it be useful to introduce some version of the WriteControl command that simply asks for a message about Dastard's writing state without asking that state to change? If the asynchronous response is dropped, one can query again and again until you get it.

ggggggggg, 2021-12-14:
Yeah that rings a bell. I like the idea of adding sync commands to just ask for some status info, pretty simple. Just an alternate info route, as long as they don't ever change any state.

joefowler, 2021-12-14:
If this is an absolutely critical requirement, I suppose we could at least ask what directory is _intended_ for writing files. But wait, is that even computed before the writing starts? No it isn't, surely. We know that all OFF files (or LJH) go into a tree anchored at `/data/tes/` or something, but the subdirectory will be `/data/tes/20211214/0003` for the 3rd run today. You can ask what subdirectory is expected, but suppose you put in the RPC request at 23:59:59, so the expected subdirectory is `20211214/0003`, but then it takes 2 seconds to start. The _actual_ subdirectory will be `20211215/0000`.

joefowler, 2021-12-14:
Re: alternate info route.

The problem is that the RPC server is a separate goroutine and doesn't have synchronous access to internal state of the other goroutines (e.g., the file writer). Without that, even read-only queries to their state mean a race condition.

joefowler, 2021-12-14:
Not a fan, but one idea is this:
* RPC server keeps track of certain critical state variables like where and whether data are being written.
* Every other goroutine that changes these critical ones has to message the RPC server on a go channel with updates.
* RPC server has to select on new RPC commands or the messages to update the "outgoing state".
* RPC replies always contain this bundle of state.

I think this will get totally out of hand. Adding a new variable to the "critical state slate" is a giant pain. Not only does the expected RPC reply packet change, but you have to communicate state updates to the server via a new channel. Yuck.

ggggggggg, 2021-12-14:
Yeah I don't like adding state to the RPC. But I think just adding a sync call that returns the current state is pretty harmless given that we already handled the race condition with our queing thing.

joefowler, 2021-12-14:
Yep. I think this is the solution. It's possible that the existing commands would solve the problem (i.e., maybe telling it to write twice has no effect other than getting the writing-state info twice?). But if not, it would be easy to add a new "tell me the writing state" command.

joefowler, 2022-03-11:
In chat with @ggggggggg we decided that this feature is not actually needed.

closed 2022-03-11 by joefowler