usnistgov/dastard #271: Dastard crashes with high memory usage if config is present

cjtitus, 2022-05-13:
I'm on emtfix2 branch, commit a0aaa86e10e9f8

When I start Dastard, then Dcom, and start data, if I have a config file present, Dastard basically runs my computer out of memory, grinds to a halt, and then dcom crashes due to missed heartbeats. Dastard itself doesn't always crash, but I need to kill it to keep it from taking out my PC.

I have attached a config file that caused a crash at startup. However, as this config file was necessarily the result of a previous, working configuration, I have no reason to actually suspect that the specific configuration is the problem. Rather, starting Dastard with any config seems to be the problem.
[config_pathological.txt](https://github.com/usnistgov/dastard/files/8688946/config_pathological.txt)

ggggggggg, 2022-05-13:
I'll look at this, should be fairly straightforward.

joefowler, 2022-05-13:
Fixed by #272.

closed 2022-05-13 by joefowler