usnistgov/dastard #285: Configuration YAML needs more explanation

joefowler, 2022-09-23:
The file `$HOME/.dastard/config.yaml` is an important configuration file, but we don't mention it in the README or anywhere else. Propose:
* Say briefly in the README that this exists, it's for persistence of state, and though humans _may_ edit it, they really should never ever need to.
* Add a similar note as a comment automatically into the top of the file ("this file is saved and read by Dastard...blahblahblah"), if possible.
* Other ideas?

joefowler, 2022-09-23:
Also the README should have more clarity about how to build and figure out which dastard you're running. It will still be easy to mess this up, but make sure we do what we can.

ggggggggg, 2022-09-23:
Maybe dastard should print out it's version/git hash on startup. I know we tried this but it often doesn't work.

I'd say we should add it to the readme, and note that a useful debugging step is to delete the file, but it's not intended to be edited by humans. I don't think we should add any info to the file itself because that sounds like more then 5 minutes of work.

joefowler, 2022-09-23:
I could never figure out how to make (get it? "make") Dastard know its own version and git hash except by inserting it using the "ldflags" argument when you run the "go build" command. That's the entire reason that Dastard has a `Makefile`. So my usual usage is
```
make build && ./dastard
```
This runs a command like
```
go build -ldflags "-X main.buildDate=2022-09-23.20:39:23.UTC -X main.githash=49573df" -o dastard cmd/dastard/dastard.go
```

Anyway, this might be a strong reason to put that procedure into the README.

PR #287 (2022-09-26): Config file clarification
Fixes #285.
commit 2fce18e0 (2022-09-26): Clarification about config.yaml file

Fixes #285.
closed 2022-09-26 by joefowler