usnistgov/dastard #289: cmd/bahama/bahama.go:240:26: invalid operation: 1 << FractionBits (shift count type int, must be unsigned integer)

DarkTyr, 2022-09-26:
Are the lines 233 to 242 even needed? As it stands they will never be executed. My proposed change is to remove lines 233 to 242 out of bahama.go

The other concern is why none else has had this issue on a fresh master git clone.

joefowler, 2022-09-26:
I don't understand this issue. 
* Is this a compile-time issue? If so, what is your result of `go version`?
* Is this a run-time issue? If so, how can it be exercised?
* What exact commit of dastard are you trying? We've made about 4-6 updates to master in the last few days, some of them today.

joefowler, 2022-09-26:
No, they are not needed, provided we never go back to the old behavior where fewer than 16 bits are used to represent a phi0. I think this is here just to make Bahama easier to update if we ever go back to that way. Worth it? Perhaps not.

DarkTyr, 2022-09-26:
compile-time issue. go1.10.4 so it needed to be updated. 
When I say fresh master git clone, I mean right before attempting to compile. Which yes, means the latest you got (a395487) tagged v0.2.14.

joefowler, 2022-09-26:
Okay, good--good on using the latest. But Dastard definitely won't work on pre-1.16 versions of Golang. I'm not sure what's the story for the version being so old. Maybe that's the default on Ubuntu 18.04?

DarkTyr, 2022-09-26:
Correct, that is the latest apt-get version.

DarkTyr, 2022-09-26:
It was the Golang version. with 1.16.15 installed it works fine. starting roach testing now.

closed 2022-09-26 by DarkTyr