usnistgov/dastard #293: sub-record length sample separations using legacy trigger

danbek, 2022-11-16:
Using dastard 0.2.11, hash a573d0d, edge triggering on the Advanced Triggering tab, I see pulse records separated by less than the record length.

I'm basing this on looking at p_rowcount/77, where here 77 is the number of channels being streamed by the FPGA.

I've attached a zip file with a python script that demonstrates the problem, and the noise file for one channel, but the pulse files was too bug to upload to github. You can find it on medusa, filename 20221115_run0002_chan6.ljh. You can grab other LJH files for other channels from there too.

[bug-dastard-trigger.zip](https://github.com/usnistgov/dastard/files/10025446/bug-dastard-trigger.zip)

joefowler, 2022-11-17:
That commit is over one year old. I can't hope to remember all the changes made in Dastard since then, but is it possible that this is a bug long since fixed?  

If you look at the [release notes for versions 0.2.12-0.2.14](https://github.com/usnistgov/dastard/releases), you'll see quite a few changes that concern triggering. Similarly, in [VERSIONS.md](https://github.com/usnistgov/dastard/blob/master/VERSIONS.md) you'll see that triggers are involved in the resolved  issues #247, #256, #259, #260, #235, #255, #271, #277, and #299.

Proposal: we update Dastard on the relevant computer (`687medusa`) and try again before we conclude that there's a bug.

joefowler, 2022-11-17:
Idea: the logic of the function `edgeTriggerComputeAppend` makes this problem impossible..._within a single DataSegment_.  

Maybe the problem happens when you cross the boundary from one segment to another? That is, maybe the next call to `edgeTriggerComputeAppend` is not properly imposing the necessary delay of 1 record length between one trigger and the next?

danbek, 2022-11-17:
After updating to DASTARD 0.2.15, hash 85ab821, we still have the problem where ~ 1 % of pulses are just delayed versions of the prior pulse.

But there were also a large number (25 %) that were exact duplicates of the prior pulse, down to the rowcount value.

LJH files are on medusa, /data/20221117/0000 for noise files, /data/20221117/0001 for the pulses.

danbek, 2022-11-18:
Using the "Triggering" tab, threshold = 100, 15 monotonic, two overlapping records, I don't see this problem at all after 1125 pulses. This is looking at channel 6.

I do see two pulses that are "delayed copies" of the prior pulse (376/377, and 732/733), but both of these are cases where a second pulse arrived. Since I had "two overlapping records" turned on, this doesn't indicate any kind of bug.

see  /data/20221117/0001. No noise records today, we don't need them for looking at this issue.

joefowler, 2022-11-18:
I think we see the problem. In dd889c46931 (March 2022), we changed how many samples are retained from one data processing "segment" for looking at when the next segment is ready. Unfortunately, the legacy trigger algorithms assume that it's safe to look at the new data starting with the first sample.

I'll make a test for this, then I'll fix it.

commit 6ee82c38 (2022-11-30): Fix unwanted overlapping legacy triggers

Fixes #293
commit 7d64d97e (2022-11-30): Merge pull request #296 from usnistgov/overlap_record_bug

Overlap record bug. Fixes #293
closed 2022-11-30 by joefowler