usnistgov/dastard #297: ROACH data source needs correct unwrapping behavior

joefowler, 2023-01-26:
The ROACH2 firmware that @DarkTyr recently wrote for the HOLMES team does not properly unwrap phases. Is this a question of turning on the unwrap feature? Or of having the wrong parameters for it? Fix it.

This might also require Dcom changes--not sure.

DarkTyr, 2023-01-28:
1. Network Byte order which is BIG ENDIAN 
2. We are outputting "Scaled Radians" from the last ARCTAN with a bit width of 16. This means that the format is "Fix16_13", 16 bits wide, 13 bits fractional. 
"""
Example from Xilinx Documentation:

Fix10_7 format values, +1 and -1 are represented as:

"0010000000" => 001.0000000 => +1.0
"1110000000" => 111.0000000 => - 1.0
"""

The annoying part about this I think is that + and - are inclusive and valid values. Technically the range is not just the lowest 13 bits. My recollection from Matter is that we treated it as a 13 bit number when unwrapping, masked off the upper bits (We put the additional sync signal in the MSB) and then fill in the top three bits with unwrapped phase.

joefowler, 2023-01-30:
You mean the full range is {xe000, e001, e002, ... ffff, 0000, 0001, ... 1ffe, 1fff, 2000}, and these cover angles from [-pi, +pi] radians, or [-180, +180] degrees?

The existing Dastard code seems like it _correctly_ understands that the lowest 14 bits are significant. It's not a problem with the number of bits. I think the problem is that it's expecting the top two bits to be 00, when in fact they are sign bits (if we ignore the single possible value of 0x2000).

If I have that correct, then the right approach would be for Dastard to mask the top 2 bits to 00 and treat the result as an unsigned 14-bit number. The effect would be to convert (say) -90 degrees to +270. Which is totally reasonable--they are equivalent anyway. This would convert angle "+1.0" to "-1.0", but again, they are already equivalent, so there's no reason not to do this.

Anyway, your recollection is right (except replace "13" with "14" in what you said). The problem that's new in Dastard is that Matter masked the upper bits and Dastard didn't. But now it will, now that I realize the problem!

joefowler, 2023-01-30:
Yep, in [ndfb_server:RoachUDPProcessor.cpp line 128](https://bitbucket.org/nist_microcal/nasa_daq/src/master/ndfb_server/RoachUDPProcessor.cpp#lines-128) you can see that it doesn't mask the top 2 bits, but instead it bit-shifts as a signed integer:
```c++
int16_t val = ((int16_t)raw_val) >> bits_to_shift;
```
That would also work. But in Dastard, raw data values are always treated as `RawType` which is unsigned: https://github.com/usnistgov/dastard/blob/8055b1cc3bbed426971835ad94d1880c37f62d27/data_source.go#L19

We can't treat signed data as unsigned and do a right shift, but that's what we have been doing. Now I see the problem. Thanks @DarkTyr.