usnistgov/dastard #314: LJH, OFF: We should make file-magic descriptions for the unix "file" command

joefowler, 2023-03-10:
If we put files in `/usr/share/file/magic/` that describe how the OS can recognize your file types for the `file` command, then we can teach the OS to say a few useful things about an LJH or OFF file.

Wish to see:
```bash
pcuser@localhost ~ $ file 20230220_run0000_chan0.ljh 
20230220_run0000_chan0.ljh: LJH pulse data version 2.2, 2500 samples/record, samples 8e-7 seconds
```

What you actually see now:
```bash
pcuser@localhost ~ $ file 20230220_run0000_chan0.ljh 
20230220_run0000_chan0.ljh: data
```

See [this example](https://sourceforge.net/p/slimdata/code/ci/master/tree/doc/magic) of a magic file I made for "slim", my data compression algorithm. Don't know if we can do this with LJH files, given how free-form their key-value specs are, but it would be cool if we can.

ggggggggg, 2023-03-10:
I think the number of people who even know about the `file` command are quite small.... but this won't hurt anything so fine.

joefowler, 2023-03-10:
I don't see how to make it work for OFF files, but this is the spirit for LJH files:
```text
# ljh:  file(1) magic for LJH pulse data files.
0     string  #LJH\ Memorial\ File\ Format    LJH pulse data
>&0   regex/4096   \^Save\ File\ Format\ Version:\  version
>>&0  regex/64     .*                               %s
>&0   regex/4096   \^Total\ Samples:\               \b,
>>&0  regex/4096   .*\$                             %s samples per record
>&0   regex/4096   \^Timebase:\                     \b, sample period
>>&0  regex/4096   .*\$                             %s sec
## End of LJH
```

joefowler, 2023-03-11:
I still think this is a cute idea, but
* I couldn't get it to detect OFF files anything other than `JSON data`
* I couldn't install the magic into `/usr/share/files/magic/` on a Mac, because `/usr/` is protected there.
* I couldn't find a workaround via something in `/usr/local/` or an environment variable. (`MAGIC` can override the system magic info, but cannot seem to _add_ to it)

I'm going to mark won't-fix and close this. Probably it would be better to write a simple LJH/OFF diagnostic tool in Python (or Go?) and make it be part of Mass or Dastard. I'll ponder that instead.

closed 2023-03-11 by joefowler