usnistgov/dastard #353: dastard crash while taking pulses

danbek, 2024-07-03:
I got this crash while taking pulse data.

It crashed within 10 minutes of starting a new pulse acquisition. But the dastard instance had been up for 24+ hours, and had been taking pulses for 23.5 hours prior to this.

Once I restarted dastard and started a new acquisition, it ran for another 10 hours without problems.

sampling rate was 488 kHz, 9 channels, channel numbers started at 12288. new-style triggering.

This was on minotaur.

```
$ dastard

This is DASTARD version 0.3.4pre1 (git commit 94a2836)
Logging problems       to /home/pcuser/.dastard/logs/problems.log
Logging client updates to /home/pcuser/.dastard/logs/updates.log

2024/06/28 16:46:32 Dastard config file: /home/pcuser/.dastard/config.yaml
2024/06/28 16:46:34 New client connection established
2024/06/28 16:46:45 Starting data source named ABACOSOURCE
Sample rate for chan [12288-12296] 488281 /sec determined from  100 packets: Ît=0.025141 sec, Îserial=99, and 124.000 samp/packet
Panic! internal buffersChan full, len 100, capacity 100
2024/06/29 16:56:22 Buffer 100/100, now-firstTime 1m28.905327339s
2024/06/29 16:56:22 Buffer 99/100, now-firstTime 1m28.878819496s
2024/06/29 16:56:22 Buffer 98/100, now-firstTime 1m28.829152868s
2024/06/29 16:56:22 Buffer 97/100, now-firstTime 1m28.779340966s
2024/06/29 16:56:22 Buffer 96/100, now-firstTime 1m28.729822433s
2024/06/29 16:56:22 Buffer 95/100, now-firstTime 1m28.680236198s
2024/06/29 16:56:22 Buffer 94/100, now-firstTime 1m28.630753628s
2024/06/29 16:56:22 Buffer 93/100, now-firstTime 1m28.58097499s
2024/06/29 16:56:22 Buffer 92/100, now-firstTime 1m28.531257027s
2024/06/29 16:56:22 Buffer 91/100, now-firstTime 1m28.481552965s
2024/06/29 16:56:22 Buffer 90/100, now-firstTime 1m28.431782716s
2024/06/29 16:56:22 Buffer 89/100, now-firstTime 1m28.382023202s
2024/06/29 16:56:22 Buffer 88/100, now-firstTime 1m28.33221489s
2024/06/29 16:56:22 Buffer 87/100, now-firstTime 1m28.28239388s
2024/06/29 16:56:22 Buffer 86/100, now-firstTime 1m28.232881515s
2024/06/29 16:56:22 Buffer 85/100, now-firstTime 1m28.183100807s
2024/06/29 16:56:22 Buffer 84/100, now-firstTime 1m28.133351679s
2024/06/29 16:56:22 Buffer 83/100, now-firstTime 1m28.083589222s
2024/06/29 16:56:22 Buffer 82/100, now-firstTime 1m28.033864492s
2024/06/29 16:56:22 Buffer 81/100, now-firstTime 1m27.984049272s
2024/06/29 16:56:22 Buffer 80/100, now-firstTime 1m27.934232063s
2024/06/29 16:56:22 Buffer 79/100, now-firstTime 1m27.8843573s
2024/06/29 16:56:22 Buffer 78/100, now-firstTime 1m27.834837368s
2024/06/29 16:56:22 Buffer 77/100, now-firstTime 1m27.784851765s
2024/06/29 16:56:22 Buffer 76/100, now-firstTime 1m27.735006731s
2024/06/29 16:56:22 Buffer 75/100, now-firstTime 1m27.685467894s
2024/06/29 16:56:22 Buffer 74/100, now-firstTime 1m27.635671369s
2024/06/29 16:56:22 Buffer 73/100, now-firstTime 1m27.585885491s
2024/06/29 16:56:22 Buffer 72/100, now-firstTime 1m27.535944027s
2024/06/29 16:56:22 Buffer 71/100, now-firstTime 1m27.486247364s
2024/06/29 16:56:22 Buffer 70/100, now-firstTime 1m27.436469678s
2024/06/29 16:56:22 Buffer 69/100, now-firstTime 1m27.386636438s
2024/06/29 16:56:22 Buffer 68/100, now-firstTime 1m27.337093333s
panic: internal buffersChan full, len 100, capacity 100

goroutine 21 [running]:
github.com/usnistgov/dastard.(*AbacoSource).readerMainLoop(0xc0000e7188)
        /home/pcuser/qsg_git_clones/dastard/abaco.go:1071 +0xdf7
created by github.com/usnistgov/dastard.(*AbacoSource).StartRun in goroutine 50
        /home/pcuser/qsg_git_clones/dastard/abaco.go:951 +0xd6
```

danbek, 2024-09-12:
It happened again today. Also while writing data. Looks like the similar / same error. Once again on minotaur.

In this case, one of the channels is going into an unstable state from time-to-time, causing it to trigger many times. Not sure whether that is contributing to the problem.

```
$ dastard

This is DASTARD version 0.3.4pre1 (git commit 94a2836)
Logging problems       to /home/pcuser/.dastard/logs/problems.log
Logging client updates to /home/pcuser/.dastard/logs/updates.log

2024/09/12 08:43:42 Dastard config file: /home/pcuser/.dastard/config.yaml
2024/09/12 08:43:46 New client connection established
2024/09/12 08:43:48 Starting data source named ABACOSOURCE
Sample rate for chan [12288-12296] 488281 /sec determined from  100 packets: Ît=0.025141 sec, Îserial=99, and 124.000 samp/packet
2024/09/12 11:28:39 Buffer 100/100, now-firstTime 43.749933159s
Panic! internal buffersChan full, len 100, capacity 100
panic: internal buffersChan full, len 100, capacity 100

goroutine 38 [running]:
github.com/usnistgov/dastard.(*AbacoSource).readerMainLoop(0xc000198e08)
        /home/pcuser/qsg_git_clones/dastard/abaco.go:1071 +0xdf7
created by github.com/usnistgov/dastard.(*AbacoSource).StartRun in goroutine 5
        /home/pcuser/qsg_git_clones/dastard/abaco.go:951 +0xd6
```

DarkTyr, 2024-09-14:
Is the location where you are saving the pulse records on the HDD or SSD? I believe the standard procedure here is to initially save data to the system SSD and then move over to the HDD when the run is complete. This is to avoid issues with buffering to disk when the random write speed can't keep up.

ggggggggg, 2024-09-16:
Also if it is an HDD, then we probably have a fix, and it would be nice to test it on that system

danbek, 2024-09-18:
Both of these happened while writing to /data on minotaur. df tells me that `/data` is `/dev/sdb3`, which has 8 TB, so I think it must be HDD.

Are you guys sure that this is a problem with buffering to disk?

It would be fine to install a fix on minotaur, but given how intermittent the problem is (happened just twice to me since July) I'm not sure that we can really "test" it that way.

Could we try to force the error by writing autotriggers for a long time?

ggggggggg, 2024-09-18:
I'm not sure, but we've seen it as BESSY, and something similar at NSLS. Basically we need to call `File.Sync()` instead of `File.Flush()` to fix NSLS, so it may fix yours as well. So I think when we fix #354 this will be fixed.